nekromant/esp8266-frankenstein #15: Handling of whitespace in arguments

Chaemelion, 2015-01-03:
From what I can tell, the command interpreter splits args on whitespace, correct? Without escaping methods, I'm unable to connect to SSIDs that contain whitespace.

nekromant, 2015-01-03:
Yep, that's a limitation of microrl. See src/microrl.c 
From what I've seen adding whitespace handling might be a little tricky, but doable. It's somewhere on my TODO list.

Chaemelion, 2015-01-04:
Gotcha. The original AT firmware appears to delimit on commas, perhaps that would provide a simple temporary fix rather than full implementation of escape sequences. When I finally get my libs, includes, and paths configured correctly, I'll play around with testing some fixes.

d-a-v, 2015-02-23:
would the latest pull request solve this issue?

nekromant, 2015-02-23:
Yeah, it does. Thanks for fixing this one, I'm closing the issue.

closed 2015-02-23 by nekromant