nekromant/esp8266-frankenstein #18: build failed.

supergis, 2015-01-05:
I am get latest sourcecode and compile today , got errors as following: 
  [LD]        images/antares.elf
esptool.py elf2image -o images/antares- images/antares.elf
usage: esptool [-h] [--port PORT]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image}
               ...
esptool: error: unrecognized arguments: -o images/antares.elf
make: *** [images/antares.rom] 错误 2

if any changed? as previous build is success.

nekromant, 2015-01-05:
Weird, looks like a problem with esptool.py. I'll give it a spin with current esptool git master.

supergis, 2015-01-05:
I checked https://github.com/themadinventor/esptool, which usage as:
Converting an ELF file to the two binary blobs to be flashed:
./esptool.py elf2image my_app.elf
This creates my_app.elf-0x00000.bin and my_app.elf-0x40000.bin.
But in arch.mk,as:
esptool.py elf2image -o $(IMAGENAME)- $(IMAGENAME).elf
may be the problem.

supergis, 2015-01-05:
Because "./esptool.py make_image " will don't fill the gap of two segment,so result bin will not run correctly. I think use esptool rather than esptool better.

nekromant, 2015-01-05:
Hm, are you sure you're running the latest antares experimental branch? For this was fixed by e4cb0d2bb5c150e8e6cae90bd481ecbf61f1cab8

supergis, 2015-01-05:
I changed arch.mk as follow,can output flash image now:
$(IMAGENAME).rom: $(IMAGENAME).elf
	#BINOUT=./../../../$(IMAGENAME).elf
	esptool -eo $(IMAGENAME).elf -bo $(IMAGENAME)-$(FW_FILE_1).bin -bs .text -bs .data -bs .rodata -bc -ec
	esptool -eo $(IMAGENAME).elf -es .irom0.text $(IMAGENAME)-$(FW_FILE_2).bin -ec
	#esptool.py elf2image -o $(IMAGENAME)- $(IMAGENAME).elf
	dd if=/dev/zero of=$(@) bs=1K count=512
	dd if=$(IMAGENAME)-$(FW_FILE_1).bin of=$(@) conv=notrunc
	dd if=$(IMAGENAME)-$(FW_FILE_2).bin of=$(@) bs=1 seek=$$(($(FW_FILE_2)))

nekromant, 2015-01-05:
That's exactly the old way, before all job was relayed to esptool.py. Just try updating antares. 
See: https://github.com/nekromant/antares/blob/e4cb0d2bb5c150e8e6cae90bd481ecbf61f1cab8/src/arch/esp8266/arch.mk

closed 2015-01-05 by nekromant