nekromant/esp8266-frankenstein #19: Can't get passed authenticating on Android

hackwin, 2015-01-27:
Android never finishes connecting.  Occurs for all AP encryption types.

nekromant, 2015-01-28:
Is dhcp server enabled on fk?

hackwin, 2015-01-28:
Yes, DHCP server is enabled.

http://imgur.com/a/FiVHe

nekromant, 2015-01-28:
It does look like a DHCP problem. Maybe they've changed something in SDK. I'll try to test it ASAP.

hackwin, 2015-02-01:
Hey, if you have a new version I can test it out.  I also tried connecting via a USB Wifi adapter on Windows 7 and had a similar problem.  I don't think it's just DHCP.  The 8266 doesn't follow the command for setting encryption type.  Everything is WPA2_PSK

CobrAlex, 2015-02-04:
(fw0.9.5) After have established the connection between my nexus 5 and the ESP8266(encrypted or not). The IP informations on my Nexus 5 not seems be updated but all work well on Windows 8. Anyone have a idea?

nekromant, 2015-02-09:
Sorry folks for the delay, the deadline at my day-job is fast approaching and I have bugtracker full of insects to squash. I hope I can get back to this soon.

hackwin, 2015-02-12:
I was able to get the Frankenstein firmware working by using 3v volts and up to 500 mA power supply.

nekromant, 2015-02-15:
Okay, so I'm closing this for now. For it works on my android cell phone. I will add the note about power supply to the README.

closed 2015-02-15 by nekromant