nekromant/esp8266-frankenstein #21: Failed build on Ubuntu 14.04

MadmanSmoker, 2015-02-15:
I am unable to build frankenstein on Ubuntu. I have used the make STANDALONE=y for building the open sdk.

However when I enter make I get a number of undefined reference errors like.
In function `spi_flash_read_status':

nekromant, 2015-02-15:
Can you post the full build log? Also which version of SDK are you using?

soyersoyer, 2015-02-15:
same on os x
sdk: esp_iot_sdk_v0.9.5_15_01_23.zip

the log:
  -e \e[1;32m[INFO]-e \e[0m      Building application code
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;32m[INFO]-e \e[0m      Building antares library code and startup
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
  -e \e[1;36m[GEN]-e \e[0m       built-in.o
echo -e "`/Volumes/ESPTools/esp8266-frankenstein/antares/scripts/parseobjs /Volumes/ESPTools/esp8266-frankenstein/build/built-in.o` \n \
	`/Volumes/ESPTools/esp8266-frankenstein/antares/scripts/parseobjs /Volumes/ESPTools/esp8266-frankenstein/build/app/built-in.o`" | while read file; do \
	echo $file; \
	xtensa-lx106-elf-objcopy --rename-section .text=.irom0.text \
		--rename-section .literal=.irom0.literal $file; \
	done
/Volumes/ESPTools/esp8266-frankenstein/build/arch/esp8266/newlib-dummies.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/main.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/stdio.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/helpers.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/microrl.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/console.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/hostname.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/env.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cb.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/telnet.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_common.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_env.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/uart.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_iw.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_uart.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_ifconfig.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_gpio.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_flash.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_listen.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_send.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_ds18b20.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_tftp.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/tftp.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/api/api_lib.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/api/api_msg.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/api/err.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/api/netbuf.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/api/netdb.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/api/netifapi.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/api/sockets.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/api/tcpip.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/app/dhcpserver.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/app/espconn.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/app/espconn_udp.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/core/def.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/core/dhcp.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/core/dns.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/core/init.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/core/mem.o
/Volumes/ESPTools/esp8266-frankenstein/build/app/lwip/core/memp.o
[... 21 lines cut ...]
	`/Volumes/ESPTools/esp8266-frankenstein/antares/scripts/parseobjs /Volumes/ESPTools/esp8266-frankenstein/build/built-in.o` \
	`/Volumes/ESPTools/esp8266-frankenstein/antares/scripts/parseobjs /Volumes/ESPTools/esp8266-frankenstein/build/app/built-in.o` \
	-nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group  -lmain -lhal -lnet80211 -lphy -lpp -lwpa -lssl -lm  -lcirom -lgcc -Wl,--end-group    -lm -T /Volumes/ESPTools/esp8266-frankenstein/antares/src/arch/esp8266/ld/eagle.app.v6.ld
xtensa-lx106-elf-gcc -o images/antares.elf /Volumes/ESPTools/esp8266-frankenstein/build/arch/esp8266/newlib-dummies.o /Volumes/ESPTools/esp8266-frankenstein/build/app/main.o /Volumes/ESPTools/esp8266-frankenstein/build/app/stdio.o /Volumes/ESPTools/esp8266-frankenstein/build/app/helpers.o /Volumes/ESPTools/esp8266-frankenstein/build/app/microrl.o /Volumes/ESPTools/esp8266-frankenstein/build/app/console.o /Volumes/ESPTools/esp8266-frankenstein/build/app/hostname.o /Volumes/ESPTools/esp8266-frankenstein/build/app/env.o /Volumes/ESPTools/esp8266-frankenstein/build/app/cb.o /Volumes/ESPTools/esp8266-frankenstein/build/app/telnet.o /Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_common.o /Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_env.o /Volumes/ESPTools/esp8266-frankenstein/build/app/uart.o /Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_iw.o /Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_uart.o /Volumes/ESPTools/esp8266-frankenstein/build/app/cmd_ifconfig.o /Volumes/ESPTo [... cut ...]
(.text+0x88): undefined reference to `SPI_read_status'
/Volumes/ESPTools/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(spi_flash.o): In function `spi_flash_read_status':
(.text+0xa8): undefined reference to `SPI_write_status'
/Volumes/ESPTools/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(spi_flash.o): In function `spi_flash_read_status':
(.text+0xc0): undefined reference to `SPI_write_status'
collect2: error: ld returned 1 exit status
make: *** [images/antares.elf] Error 1

nekromant, 2015-02-15:
Updating frankenstein and antares to the latest git. I've just pushed changes to fix compilation with the latest 0.9.5 SDK

closed 2015-02-15 by nekromant

MadmanSmoker, 2015-02-15:
Yep. That works. What did you change? A git log doesnt show me any commit that updates it to version 0.9.5?

nekromant, 2015-02-15:
Everything regarding SDK happens in 'antares' tree, since it handles most of the lowlevel stuff including LD file, so have a look at the git commit log in antares. e.g.
```
cd antares
git log
```

MadmanSmoker, 2015-02-16:
Okay. Thanks for the help.