nekromant/esp8266-frankenstein #23: Multiple ds18b20 reading wanted

dvv, 2015-02-19:
Namely, the subj.
A list of `id<TAB>value\n` of all ds18b20 devices on the net would be great. Or, how it can be done otherwise?
TIA,
--Vladimir

nekromant, 2015-02-19:
Hm, I guess it might be quite easy to hack, see src/cmd_ds18b20.c to accept multiple arguments. Or just issue the command several times for each of your sensors.

darkbyte-ru, 2015-03-23:
Done by pull request #48. Example of output:
```
frankenstein > ds18b20 2
Temperature: +28.12 Celsius

frankenstein > ds18b20 2 all
2880a9040500000f +30.56
2848cdd50400001a +29.81
283a36eb04000067 +29.68
2841f1040500000c +30.0
28d57d0505000004 +29.81
286d5f0405000056 +29.81
28bd5fec0400006c +29.75
28b0a7ea040000b7 +29.50
28947ed60400002a +30.56
284c3006050000ae +29.50
28db018c040000ba +30.31
```

nekromant, 2015-03-23:
Since it's fixed by #48 by @darkbyte-ru - closing it as fixed.

closed 2015-03-23 by nekromant