nekromant/esp8266-frankenstein #24: MQTT support

Jeshwanth, 2015-02-22:
Hi, how about MQTT in Frankeinstein ? 
We can have basic envt setup in command line for MQTT. I am interested in contributing for MQTT. 
Any comments or guidelines ?

ystrem, 2015-02-22:
Hi, that would by great to have mqqt in frankeistein, but my c skill is not good enough to implement such thing. So any help would by superior. 😊

Jeshwanth, 2015-02-22:
Same here, but I am thinking of give a try :). 
Can we use this library https://github.com/tuanpmt/esp_mqtt ? 
We can have MQTT start/stop to control GPIO and send GPIO info every 1  minutes (configurable).

nekromant, 2015-02-22:
Well, I've never had a chance to give MQTT a try, so it will take someone familiar with the beast to hack in the support for it. Basically all you need is: 
1. Add all the library sources to frankenstein. You're better off starting with FK, and after it's cleaned up it can be later merged into antares, so that other architectures it supports can benefit from it.  You can safely remove ICACHE_FLASH_ATTR from all the functions with sed - FK relocates code there with no such hackery ever needed. I test it on other architectures supported by antares (e.g. stm32, avr) before merging to antares master.  
2. Add a command that will interact with MQTT library. See src/cmd_* as a reference.
3.  Remember to add enable/disable of MQTT stuff via kconfig (see kcnf file in top-level of the project)
4. If you need to start mqtt at boot - see src/main.c. user_init() is right there. 

Right now my plans (apart from sorting out a huuuge deadline at my dayjob) are: 
1. A filesystem (romfs-compatible, compressed with heatshrink) and (later) a web-server on top of that.
2. heatshrink-compression for OTA updates.  
3. Lua. I need it into antares for a different project, but FK can benefit from that as well.

Jeshwanth, 2015-02-22:
Thanks Andrew, this is best info to start :).

ystrem, 2015-03-20:
Hi, any progress in this topic ? Thanks

Jeshwanth, 2015-03-20:
I am currently working on it.
Integrated the library and testing MQTT with local broker, but its not working it. Hope will fix soon. 

Here is the repositary .
https://github.com/Jeshwanth/esp8266-frankenstein

Jeshwanth, 2015-03-20:
I am making it work now so not considered many things. Will submit it once everything works.

ystrem, 2015-03-20:
Great work, I'll take a look at it. Thanks 😊

Jeshwanth, 2015-03-24:
Hi @nekromant,

I removed all the ICACHE_FLASH_ATTR as I got below error, but still I am finding the same. Any suggestions ?

```
  [GEN]       built-in.o
  [GEN]       built-in.o
  [INFO]      Building antares library code and startup
  [CC]        newlib-dummies.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [IROMIFY]   Moving application code to IROM
  [LD]        images/antares.elf
/home/jeshwanth/mywork/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: images/antares.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
/home/jeshwanth/mywork/esp8266/FK_new/esp8266-frankenstein/antares/toolchains/gcc.mk:158: recipe for target 'images/antares.elf' failed
make: *** [images/antares.elf] Error 1
```

Thanks

nekromant, 2015-03-24:
@Jeshwanth Weird. Try a make mrproper && make build, see if things change. Something explicitly places a LOT of code into iram_1_0_seg. Are you building vanilla master or you have your own code inside?

susisstrolch, 2015-03-24:
Did you check the various
CONFIG_LIB_HEADSHRINK* settings in .config?

I have the same behaviour when f.e. CONFIG_LIB_HEATSHRINK_DECODE isn't set.

nekromant, 2015-03-24:
@susisstrolch They would go in flash, not ram. @Jeshwanth - make sure ESP8266_LIBC_IROM and ESP8266_FORCE_IROM are both enabled, they are in 'Target Platform Settings"

Jeshwanth, 2015-03-25:
@nekromant ,
The SDK 0.9.2 only right ? 

Thanks

nekromant, 2015-03-25:
@Jeshwanth It was 0.9.6 a few weeks ago, now we've already bumped to version 1.0 and are fighting all the new bugs that came along.

Jeshwanth, 2015-03-26:
@nekromant ,
Fine then, I will also try fixing bugs and continue MQTT after that :).

Jeshwanth, 2015-04-01:
@nekromant 
Finally ESP8266_LIBC_IROM and ESP8266_FORCE_IROM these two fixed the build.. Thanks a lot :)

Jeshwanth, 2015-04-01:
Finally integrated mqtt library to FK and its working :). Lot of code changes required. Will do all the cleanup and send pull request :)

hpeyerl, 2015-05-05:
Where are you on this?  I was just about to start on MQTT-SN when I saw this... I see it's been a while so before I start playing with it, do you have any local changes?

Jeshwanth, 2015-05-05:
Hi @hpeyerl , I have local changes and created command for mqtt_start, mqtt_stop, mqtt_pub and mqtt_sub. And all are tested, and nowadays I am not getting time to fix other problems. 

If you need my local changes I will put it in my github repository by tomorrow.

hpeyerl, 2015-05-06:
ok. I'll start trying to merge your changes into my tree and play with it in the meantime.

Jeshwanth, 2015-05-06:
Hi @hpeyerl , Please download v2 from https://drive.google.com/folderview?id=0BxI0oWeNLN5JfjgxS0RsSC1HX0pQT25XR3lMYllGSkVON2pyODY0ZlZQR1hTOV9oTlF5NGs&usp=sharing 

I didnt get time to upload to my repo. You can change values in cmd_mqtt.c for the broker. 

I will start again from next week,

Thanks

Jeshwanth, 2015-05-06:
BTW I used sdk version 1.0.0

hpeyerl, 2015-06-03:
Just an update ...

After a bit of a hiatus of PCB layout and board bringup, I'm back on MQTT.  I've cleaned up warnings, redid sysCfg, and am working on making the code more robust as well as binding it to devices...  I don't have anything to push yet.  If anyone else is working on this or interested in doing so, drop me a note and I can do a checkpoint push.

Jeshwanth, 2015-06-04:
@hpeyerl 
Great to know this, I am not finding much time to contribute further to this topic dou to other work commitments. Will join you in sometimes. :)

hpeyerl, 2015-06-25:
I think I have the MQTT stuff somewhat beaten into shape ... It's living on my fork for now:

https://github.com/hpeyerl/esp8266-frankenstein

I'd be happy if interested parties played with it a bit and fixed any bugs or added enhancements... 

I struggled for a while to have startmqtt delay until WIFI was connected... Right now, startup is delayed for a few TCP timeouts while WIFI is auto-connecting.   

Another issue is that 'gpio_sub' won't work until MQTT is connected so I haven't managed to get it to work automatically on powerup.   

Also, for some reason, once I publish an event via Mosquitto, the event continues to be received by FK-MQTT... I need to dig into that...

Anyway ..... I am able to get FK to publish my temperature/humidity, and subscribe to relay events so I can turn lights or my irrigation valves on/off..

I've added a command "every":

every      - Execute console cmd every N seconds
             every 10 'gpio out 5 1 ; gpio out 5 0'
             every 10 $envvar

 which can be used to regularly publish an event like so:

setenv pubs 'gpio_pub 5;gpio_pub 4'
every 15 $pubs

I've also added commands:

gpio_pub   - Publish GPIO state
             gpio_pub <gpio>
gpio_sub   - Subscribe to GPIO
             gpio_sub <gpio>

si7020_pub_humidity - Publish Humidity via MQTT
si7020_pub_temperature - Publish Temperature via MQTT

There are also commands:

mqttsub    - Subscribe MQTT
mqttpub    - Publish MQTT
stopmqtt   - Stop MQTT Service
startmqtt  - Start MQTT Service

Environment variables are how MQTT is configured:

mqtt_device_id  RelayPhrob1
mqtt_host   192.168.37.30
mqtt_user  myusername
mqtt_pass mypasswd

In $bootcmd I have:

pubs        gpio_pub 5
bootcmd     startmqtt;every 15 $pubs

Various outputs
------------------------

From Mosquitto:

$ mosquitto_pub -t "RelayPhrob1/gpio/5" -m "1"

Office/si7020/temperature/0 30
Office/si7020/humidity/0 24
RelayPhrob1/gpio/5 0
RelayPhrob1/gpio/5 1
RelayPhrob1/gpio/5 1
Office/si7020/temperature/0 30
Office/si7020/humidity/0 24
RelayPhrob1/gpio/5 1

$ mosquitto_pub -t "RelayPhrob1/gpio/5" -m "0"

RelayPhrob1/gpio/5 0
RelayPhrob1/gpio/5 0
Office/si7020/temperature/0 30
Office/si7020/humidity/0 24
RelayPhrob1/gpio/5 0

Frankenstein console:

frankenstein > gpio_pub 5
MQTT: queuing publish, length: 23, queue size(0/2048)

frankenstein > MQTT: Sending, type: 3, id: 0000
Receive topic: RelayPhrob1/gpio/5, data: 0 

MQTT: Send keepalive packet to 192.168.37.30:1883!
MQTT: Sending, type: 12, id: 0000

frankenstein > gpio_pub 5
MQTT: queuing publish, length: 23, queue size(0/2048)

frankenstein > MQTT: Sending, type: 3, id: 0000
Receive topic: RelayPhrob1/gpio/5, data: 0 
Receive topic: RelayPhrob1/gpio/5, data: 1 

frankenstein > gpio_pub 5
MQTT: queuing publish, length: 23, queue size(0/2048)

frankenstein > MQTT: Sending, type: 3, id: 0000

hpeyerl, 2015-06-25:
One additional item of note ... The code is fairly large and quickly exceeds the size of irom0.. So you have to choose either MQTT I2C or MQTT GPIO ... That's ok for my hardware since I have separate boards for relays and temp/humidity...

nekromant, 2017-12-14:
Since it looks like it's all (mostly) merged, I'm closing this issue.

closed 2017-12-14 by nekromant