nekromant/esp8266-frankenstein #25: undefined reference to `os_printf_plus'

meeusr, 2015-02-22:
Hello

after pulling the latest changes I get following error while building the firmware:
...
  [GEN]       built-in.o
  [IROMIFY]   Moving application code to IROM
  [LD]        images/antares.elf
/repo/meeusr/esp8266/esp8266-frankenstein/build/app/lwip/app/dhcpserver.o: In function `handle_dhcp':
dhcpserver.c:(.irom0.text+0x710): undefined reference to `os_printf_plus'
dhcpserver.c:(.irom0.text+0x747): undefined reference to `os_printf_plus'
/repo/meeusr/esp8266/esp8266-frankenstein/build/app/lwip/app/espconn.o: In function `espconn_accept':
espconn.c:(.irom0.text+0x4fe): undefined reference to `os_printf_plus'
/repo/meeusr/esp8266/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o: In function `espconn_server_recv':
espconn_tcp.c:(.irom0.text+0x9b6): undefined reference to `os_printf_plus'
/repo/meeusr/esp8266/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o: In function `espconn_tcp_sent':
espconn_tcp.c:(.irom0.text+0xde7): undefined reference to `os_printf_plus'
/repo/meeusr/esp8266/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o:espconn_tcp.c:(.irom0.text+0xe0a): more undefined references to `os_printf_plus' follow
collect2: error: ld returned 1 exit status
make: *** [images/antares.elf] Error 1

After changing the "if 0" in ./antares/src/arch/esp8266/include-sdk/espmissingincludes.h into "if 1", the problem is resolved.

Can I have some more background on why I need to make this change?

Regards,
Ronny

ystrem, 2015-02-22:
I made fresh clone, changed sdk version to 0.9.2 in esp-open-sdk Makefile, then tried make a got this.

```
  [IROMIFY]   Moving application code to IROM
  [LD]        images/antares.elf
/home/ystrem/esp8266-frankenstein/build/app/cmd_at.o:(.irom0.text+0x4): undefined reference to `system_get_sdk_version'
/home/ystrem/esp8266-frankenstein/build/app/cmd_at.o: In function `do_at':
cmd_at.c:(.irom0.text+0x21): undefined reference to `system_get_sdk_version'
/home/ystrem/esp8266-frankenstein/build/app/cmd_adc.o:(.irom0.text+0x4): undefined reference to `system_adc_read'
/home/ystrem/esp8266-frankenstein/build/app/cmd_adc.o: In function `do_adc':
cmd_adc.c:(.irom0.text+0x14): undefined reference to `system_adc_read'
/home/ystrem/esp8266-frankenstein/build/app/lwip/app/dhcpserver.o: In function `handle_dhcp':
dhcpserver.c:(.irom0.text+0x710): undefined reference to `os_printf_plus'
dhcpserver.c:(.irom0.text+0x747): undefined reference to `os_printf_plus'
/home/ystrem/esp8266-frankenstein/build/app/lwip/app/espconn.o: In function `espconn_accept':
espconn.c:(.irom0.text+0x4fe): undefined reference to `os_printf_plus'
/home/ystrem/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o: In function `espconn_server_recv':
espconn_tcp.c:(.irom0.text+0x9b6): undefined reference to `os_printf_plus'
/home/ystrem/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o: In function `espconn_tcp_sent':
espconn_tcp.c:(.irom0.text+0xde7): undefined reference to `os_printf_plus'
/home/ystrem/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o:espconn_tcp.c:(.irom0.text+0xe0a): more undefined references to `os_printf_plus' follow
collect2: error: ld returned 1 exit status
make: *** [images/antares.elf] Error 1
```

nekromant, 2015-02-22:
You'll need that to work in the latest (0.9.5) SDK now. Some weird hackery with printf's going on there (Sigh!).

closed 2015-02-22 by nekromant

ystrem, 2015-02-22:
Yeah, with 0.9.5 it's ok.