nekromant/esp8266-frankenstein #33: DHT11 sensor

pastcompute, 2015-03-17:
Hi,

I have written some code to read a DHT-11 sensor

https://github.com/pastcompute/esp8266-frankenstein/commit/b5343ebcd7c9eab46472c20e577f6bd7f770fe8d

I haven't made a pull request yet because I haven't had a chance to test on the hardware, but figured you may as well have a look.
I derived the algorithm from https://github.com/markruys/arduino-DHT

nekromant, 2015-03-17:
Unfortunately I have no such hardware to test myself. Maybe someone around can help out with testing?

ystrem, 2015-03-17:
Hi, if you send me the code I can test it.

nekromant, 2015-03-17:
@ystrem I've added the code to master, marked as 'untested'. See if it works for you. If it does, I'll remove 'untested'.

pastcompute, 2015-03-18:
Just testing now, my code is broken (at least for the case with nothing connected) it is not returning 'not found') so need to fix that

ystrem, 2015-03-18:
I didn't had time yesterday, I'll test it this evening.

pastcompute, 2015-03-18:
It wont work, there are bugs in the retry loop, I used an unsigned for a negative (duh)
I am trying to fix it now (at least, have it detect nothing connected... if I still have time then I'll plug something in) I'll make sure I upload whatever I end up with

pastcompute, 2015-03-18:
My latest work is in https://github.com/pastcompute/esp8266-frankenstein/tree/dht11_branch

This now correctly passes through the logic and decides nothing is connected, and prints the temperature properly.
Now my only problem is it thinks there is nothing connected when I have a DHT-11 actually connected...

pastcompute, 2015-03-18:
I need to measure elapsed time, I suspect calling os_delay(1-2useconds) is not accurate.
Do you know if there is an API for getting number of microseconds / ticks since power up?

pastcompute, 2015-03-18:
oops

nekromant, 2015-03-18:
You can use ets_timer_*, and do it the async way. That's the only exposed timer API I know of.

pastcompute, 2015-03-18:
hmm. the code I started I tested and it worked on Arduino (http://www.github.com/markruys/arduino-DHT)
So just now I ended up going back to first principles and looking at how DHT-11 actually works...
there are nice diagrams in http://embedded-lab.com/blog/?p=4333 and after reading that I dont understand how the arduino one actually worked at all because it doesn't seem to do the high/low at the start the same way...

I think I will redo the read loop from scratch; looking further down in http://embedded-lab.com/blog/?p=4333 there is a ReadByte() method that doesn't use a timer which seems to make sense.

susisstrolch, 2015-03-18:
Just tried most current commit (c39b8e554c0ddf7932dc26a5bb8ac9c250b41970)
The "dht11 4", which should query my dht22 results in:
Command dht11 takes a maximum of 1 args, 2 given

pastcompute, 2015-03-18:
Sorry, thats my fault. I changed the number of arguments because presently the sensor must be hard-coded to GPIO2.  I'll update the help to indicate this.

(I think there will need to be a gpio support library to make it easier to change gpio easily.)

Also, for the moment it probably wont work with a DHT22.
If you have one, do you have a reference to anything like  http://embedded-lab.com/blog/?p=4333  on how to use it? I know the start up sequence is different but I don't have one to test it on. I rewrote the code to work like that link it should not be too hard to add DHT-22 support back in. I guess I can go digging back into various Arduino code and try again

susisstrolch, 2015-03-19:
No worry...
I'll have a look at the DHT22 code.
Examples:
https://bitbucket.org/0ff/esp-driver-dht/src/e61f237f32eb14f6ce6f6c4feba7c5ca7299f96c/dht_sensor.c?at=master
https://github.com/fasmide/esp_dht22/blob/master/user/user_main.c

nekromant, 2015-03-19:
> (I think there will need to be a gpio support library to make it easier to change gpio easily.)

@pastcompute That should be straightforward to implement, the register map is simple. Antares itself has no arduino-like GPIO abstraction to be hooked for a reason. Unlike dealing with more complex interfaces where the impact is little any abstraction slows GPIO access a great deal. In case of arduino - 800x or so, tests were on the web a while ago.

pastcompute, 2015-03-20:
@nekromant wrt GPIO, I added https://github.com/nekromant/esp8266-frankenstein/issues/38 to track work on that

pastcompute, 2015-03-23:
@darkbyte-ru has added a DHT11 + DHT22 driver so I'll try and test that, and if it works I can remove the dht11 only driver

pastcompute, 2015-03-28:
I just tested with DHT11 and seems to work OK.
I suspect the sensor itself may not be suitable for my needs the humidity reading is not very stable (each reading changes up/down by up to 10% although stays around the same point)

PR #68 (2015-03-28): Remove duplicating functionalist of dht11 command (use dht22 now)
This will close #33 (remove duplication of dht11 device now implemented by dht22 command)
closed 2015-03-28 by pastcompute