nekromant/esp8266-frankenstein #38: Handling configurable GPIO

pastcompute, 2015-03-18:
Hi,

BTW, awesome job putting this firmware together! The whole build and flash just worked so easily, compared to some other systems I have used.

It seems various devices are hard-coded to GPIO-2, which is fine because it makes life easier getting at least something working. After looking at this it seems the problem with easily accepting a user chosen GPIO is that for some reason, the GPIO_INPUT/OUTPUT_DIS/etc functions take a numeric GPIO number but before doing that you need to call those other GPIO MUX functions with various constants the value of which seem to bear no relation to the actual GPIO number.

I think we can easily solve this by having some kind of library function that takes the GPIO number and has a big switch statement and/or uses a lookup table to do that work without it having to be repeated in each driver. I had a look at the GPIO handling in node_mcu it does something similar.

On the weekend I should have some time where I can have a go at doing this and retrofitting into at least the ds18b20 and dht11 modules as a test, do you have any preferences on where best to put this code? e.g. naming conventions, etc etc

d-a-v, 2015-03-19:
On mer., mars 18, 2015 at 03:55:32 -0700, Andrew McDonnell wrote:
> Hi,
>
> BTW, awesome job putting this firmware together! The whole build and flash just worked so easily, compared to some other systems I have used.
>
> It seems various devices are hard-coded to GPIO-2, [...]

yes that would be great !
other versions of the esp8266 have lots of gpio available on the board.

> do you have any preferences on where best to put this code? e.g. naming conventions, etc etc

I personally have no precise idea. For what it's worth, people
are used to arduino wire library (which is ported to raspi with
wiringPi).  That might be a good start.

pastcompute, 2015-03-20:
_(Carrying on from #33)_
> >    (I think there will need to be a gpio support library to make it easier to change gpio easily.)

> @pastcompute That should be straightforward to implement, the register map is simple. Antares itself has no arduino-like GPIO abstraction to be hooked for a reason. Unlike dealing with more complex interfaces where the impact is little any abstraction slows GPIO access a great deal. In case of arduino - 800x or so, tests were on the web a while ago.

I discovered pin_map.h and pin_map.c which only seems to be used by i2c, is this an attempt to do something similar?

pastcompute, 2015-03-20:
Actually it looks like it, just the i2c code is confusing because it sets a variable pinSDA but then uses a macro to refer to the same variable

Typical usage appears like it would be:

```
int gpio_num = 2;
PIN_FUNC_SELECT(pin_mux[gpio_num], pin_func[sda]);
PIN_PULLDWN_DIS(pin_mux[gpio_num]);
PIN_PULLUP_DIS(pin_mux[gpio_num]);
GPIO_DIS_OUTPUT(gpio_num);
// etc
```

So someone has already done it

What I might do is tidy up the pin_mux files and document it a bit

nekromant, 2015-03-20:
I guess this would be the work of @darkbyte-ru, since the whole i2c subsystem is his work.

darkbyte-ru, 2015-03-20:
pin_map is a part of nodumcu project, so this picture will be describe black magic of pin_map.
[image]

Currently most of code for esp8266 looks like mess, I think because of poor SDK. Anyway, i2c library needs a bit deeper refactoring. Maybe there is would be better to not use such mapping? Good point of using it is much easier check for gpio pin params (unsigned <GPIO_PIN_NUM), but bad point is non logic mapping for thouse, who has no nodemcu dev board.

darkbyte-ru, 2015-03-20:
How about this:

```
uint32_t pin_mux[GPIO_PIN_NUM] = {
	PERIPHS_IO_MUX_GPIO0_U, PERIPHS_IO_MUX_U0TXD_U, PERIPHS_IO_MUX_GPIO2_U, PERIPHS_IO_MUX_U0RXD_U,
	PERIPHS_IO_MUX_GPIO4_U, PERIPHS_IO_MUX_GPIO5_U, -1, -1, 
	-1, PERIPHS_IO_MUX_SD_DATA2_U, PERIPHS_IO_MUX_SD_DATA3_U, -1, 
	PERIPHS_IO_MUX_MTDI_U, PERIPHS_IO_MUX_MTCK_U, PERIPHS_IO_MUX_MTMS_U, PERIPHS_IO_MUX_MTDO_U, 
	PAD_XPD_DCDC_CONF
}

uint8_t pin_func[GPIO_PIN_NUM] = {
	FUNC_GPIO0, FUNC_GPIO1, FUNC_GPIO2, FUNC_GPIO3, 
	FUNC_GPIO4, FUNC_GPIO5, -1, -1, 
	-1, FUNC_GPIO9, FUNC_GPIO10, -1, 
	FUNC_GPIO12, FUNC_GPIO13, FUNC_GPIO14, FUNC_GPIO15, 
	-1
};
```

Not too bad, i think. And not too complex to check for invalid gpio pin.

nekromant, 2015-03-20:
@darkbyte-ru Regarding i2c, don't refactor too much just yet - I was thinking about a generic i2c framework with bitbanging support for antares. I hope it will hit antares experimental by the end of this month. When it's done - we can clean things up.

pastcompute, 2015-03-20:
@nekromant I might close the pull request I made, because assumptions I made are obviously incorrect (although they make the current mapping work it is using a non-applicable abstraction)

What I could do is make a new file gpio_map.h using @darkbyte-ru suggestion above, but I might choose a different name, say, `gpio_pin_mux[]` and `gpio_pin_func[]`; 
For the moment, then , i2c would remain unchanged internally, but could have a new local mapping in `do_i2c()` to translate esp8266 gpio to the legacy pin_map code

I could probably wrap the mappings in a macro so if things change underneath in the future we wont have to change everywhere. I'll use dht11 and ds18b20 as proof of concept

pastcompute, 2015-03-20:
Actually you already merged it, so what I'll do is make a macro for use by all the modules, that wraps pin_map so that when things get refactored like you say, the macros wont need to change...

darkbyte-ru, 2015-03-20:
@nekromant, ok, I will only remove pin_num array and change pin_mux and pin_func, so i2c pin mapping will have the same mapping as gpio command.

pastcompute, 2015-03-20:
Ok, I might do this in the future then, I dont want to accidentally break your code and I am not in a position to regression test it easily

darkbyte-ru, 2015-03-23:
Sorry for long time waiting, I make pull request with conflicts, so it's closed now and new request created: #48.

pastcompute, 2015-03-23:
@darkbyte-ru Hey, awesome stuff!

I'll put the documentation back now, it should be shorter and easier too without all the nodemcu wierdness

pastcompute, 2015-03-23:
I guess I can close this now... thanks!

closed 2015-03-23 by pastcompute