nekromant/esp8266-frankenstein #39: Can't compile

dvv, 2015-03-19:
```
vd@VD:~/projects/esp8266/esp8266-frankenstein$ uname -a
Linux VD 3.16.0-29-generic #39-Ubuntu SMP Mon Dec 15 22:28:14 UTC 2014 i686 i686 i686 GNU/Linux
```

Current master
```
c39b8e5 Merge pull request #37 from pastcompute/dht11_works
```

SDK 0.9.5

```
src/arch/esp8266/include-sdk/user_interface.h:11,
                 from /home/vd/projects/esp8266/esp8266-frankenstein/src/console.c:7:
/home/vd/projects/esp8266/esp8266-frankenstein/include/console.h:31:26: error: 'ets_uart_printf' undeclared here (not in a function)
 #define SERIAL_PRINTF    ets_uart_printf
                          ^
/home/vd/projects/esp8266/esp8266-frankenstein/src/console.c:18:27: note: in expansion of macro 'SERIAL_PRINTF'
 printf_f console_printf = SERIAL_PRINTF;
                           ^
/home/vd/projects/esp8266/esp8266-frankenstein/antares/make/Makefile.build:96: ошибка выполнения рецепта для цели «console.o»
make[1]: *** [console.o] Ошибка 1
/home/vd/projects/esp8266/esp8266-frankenstein/antares/toolchains/gcc.mk:119: ошибка выполнения рецепта для цели «builtin»
make: *** [builtin] Ошибка 2
vd@VD:~/projects/esp8266/esp8266-frankenstein$ 
vd@VD:~/projects/esp8266/esp8266-frankenstein$ 
vd@VD:~/projects/esp8266/esp8266-frankenstein$ 

vd@VD:~/projects/esp8266/esp8266-frankenstein$ make
  [INFO]      Building application code
  [CC]        main.o
  [CC]        helpers.o
  [CC]        microrl.o
/home/vd/projects/esp8266/esp8266-frankenstein/src/microrl.c: In function 'microrl_set_prompt':
/home/vd/projects/esp8266/esp8266-frankenstein/src/microrl.c:34:2: warning: implicit declaration of function 'ets_sprintf' [-Wimplicit-function-declaration]
  ets_sprintf(current_prompt, "\n%s > ", prompt);
  ^
  [CC]        console.o
/home/vd/projects/esp8266/esp8266-frankenstein/src/console.c:18:27: error: 'SERIAL_PRINTF' undeclared here (not in a function)
 printf_f console_printf = SERIAL_PRINTF;
                           ^
/home/vd/projects/esp8266/esp8266-frankenstein/antares/make/Makefile.build:96: ошибка выполнения рецепта для цели «console.o»
make[1]: *** [console.o] Ошибка 1
/home/vd/projects/esp8266/esp8266-frankenstein/antares/toolchains/gcc.mk:119: ошибка выполнения рецепта для цели «builtin»
make: *** [builtin] Ошибка 2
vd@VD:~/projects/esp8266/esp8266-frankenstein$ 

vd@VD:~/projects/esp8266/esp8266-frankenstein$ make
  [INFO]      Building application code
  [CC]        console.o
/home/vd/projects/esp8266/esp8266-frankenstein/src/console.c:19:27: error: 'ets_uart_printf' undeclared here (not in a function)
 printf_f console_printf = ets_uart_printf;
                           ^
/home/vd/projects/esp8266/esp8266-frankenstein/antares/make/Makefile.build:96: ошибка выполнения рецепта для цели «console.o»
make[1]: *** [console.o] Ошибка 1
/home/vd/projects/esp8266/esp8266-frankenstein/antares/toolchains/gcc.mk:119: ошибка выполнения рецепта для цели «builtin»
make: *** [builtin] Ошибка 2
vd@VD:~/projects/esp8266/esp8266-frankenstein$ 

vd@VD:~/projects/esp8266/esp8266-frankenstein$ make
  [INFO]      Building application code
  [CC]        main.o
  [CC]        helpers.o
  [CC]        microrl.o
/home/vd/projects/esp8266/esp8266-frankenstein/src/microrl.c: In function 'microrl_set_prompt':
/home/vd/projects/esp8266/esp8266-frankenstein/src/microrl.c:34:2: warning: implicit declaration of function 'ets_sprintf' [-Wimplicit-function-declaration]
  ets_sprintf(current_prompt, "\n%s > ", prompt);
  ^
  [CC]        console.o
[... 144 lines cut ...]
  [INFO]      Building application code
  [CC]        main.o
  [CC]        stdio.o
  [CC]        helpers.o
  [CC]        microrl.o
/home/vd/projects/esp8266/esp8266-frankenstein/src/microrl.c: In function 'microrl_set_prompt':
/home/vd/projects/esp8266/esp8266-frankenstein/src/microrl.c:34:2: warning: implicit declaration of function 'ets_sprintf' [-Wimplicit-function-declaration]
  ets_sprintf(current_prompt, "\n%s > ", prompt);
  ^
  [CC]        console.o
In file included from /home/vd/projects/esp8266/esp8266-frankenstein/include/main.h:5:0,
                 from /home/vd/projects/esp8266/esp8266-frankenstein/include/lwip-esp8266/lwip/opt.h:47,
                 from /home/vd/projects/esp8266/esp8266-frankenstein/include/lwip-esp8266/lwip/ip_addr.h:35,
                 from /home/vd/projects/esp8266/esp8266-frankenstein/antares/src/arch/esp8266/include-sdk/user_interface.h:11,
                 from /home/vd/projects/esp8266/esp8266-frankenstein/src/console.c:7:
/home/vd/projects/esp8266/esp8266-frankenstein/include/console.h:31:26: error: 'ets_uart_printf' undeclared here (not in a function)
 #define SERIAL_PRINTF    ets_uart_printf
                          ^
/home/vd/projects/esp8266/esp8266-frankenstein/src/console.c:18:27: note: in expansion of macro 'SERIAL_PRINTF'
 printf_f console_printf = SERIAL_PRINTF;
                           ^
```

How can I fix that?
TIA

dvv, 2015-03-19:
config attached
```
#
# Automatically generated file; DO NOT EDIT.
# Antares Firmware Config
#

#
# Target platform settings
#
# CONFIG_ARCH_8051 is not set
# CONFIG_ARCH_ARM is not set
# CONFIG_ARCH_AVR is not set
CONFIG_ARCH_ESP8266=y
# CONFIG_ARCH_GBZ80 is not set
# CONFIG_ARCH_MIPS is not set
# CONFIG_ARCH_MSP430 is not set
# CONFIG_ARCH_NATIVE is not set
# CONFIG_ARCH_PIC32 is not set
CONFIG_ESP8266_FORCE_IROM=y
CONFIG_ESP8266_LIBC_IROM=y
# CONFIG_ESP8266_NEED_SDK_PATH is not set

#
# Configure libraries to link against
#
CONFIG_ESP8266_BLOB_MAIN=y
CONFIG_ESP8266_BLOB_HAL=y
CONFIG_ESP8266_BLOB_NET80211=y
CONFIG_ESP8266_BLOB_PHY=y
CONFIG_ESP8266_BLOB_PP=y
CONFIG_ESP8266_BLOB_WPA=y
CONFIG_ESP8266_BLOB_SSL=y
# CONFIG_ESP8266_BLOB_LWIP is not set

#
# Toolchain settings
#
# CONFIG_SHOW_ALL_TOOLCHAINS is not set
CONFIG_TOOLCHAIN_GCC=y
CONFIG_TOOLCHAIN_PREFIX="xtensa-lx106-elf-"
# CONFIG_CC_OPT0 is not set
# CONFIG_CC_OPT1 is not set
# CONFIG_CC_OPT2 is not set
# CONFIG_CC_OPT3 is not set
CONFIG_CC_OPTSZ=y
# CONFIG_GCC_NOSTDLIBS is not set
# CONFIG_GCC_G is not set
# CONFIG_GCC_LC is not set
CONFIG_GCC_LM=y
# CONFIG_GCC_FPIC is not set
# CONFIG_GCC_STRIP is not set
# CONFIG_GCC_PARANOID_WRN is not set

#
# Fine tuning
#
CONFIG_CFLAGS=""
CONFIG_ASFLAGS=""
CONFIG_LDFLAGS=""
CONFIG_ELFFLAGS=""

#
# Libraries and drivers
#
CONFIG_SHOW_BROKEN=y
# CONFIG_LIB_ANTARES_CORE is not set

#
# Console and IO
#
# CONFIG_LIB_EARLYCON is not set
# CONFIG_LIB_PRINTK is not set
CONFIG_LIB_PANIC_NONE=y
# CONFIG_LIB_PANIC is not set
# CONFIG_LIB_NLIBSTUBS is not set

#
# Wireless devices
#
# CONFIG_LIB_RF24 is not set

#
# Data storage devices
#
# CONFIG_LIB_SPISD is not set

#
# Input devices
#
# CONFIG_LIB_CAPSENSE is not set

#
# Misc drivers and libraries
#
# CONFIG_LIB_XSSCU is not set

#
# Data transfer protocols
#
# CONFIG_LIB_XMODEM is not set
# CONFIG_LIB_DELAY is not set
# CONFIG_LIB_STLINKY is not set
# CONFIG_LIB_URPC is not set

#
# 3rd-party libraries
#
# CONFIG_CONTRIB_LWIP is not set

#
# Project Settings
#

#
# Modules to build
#
CONFIG_CMD_IW=y
CONFIG_CMD_UART=y
CONFIG_CMD_IF=y
CONFIG_CMD_GPIO=y
CONFIG_CMD_FLASH=y
CONFIG_CMD_LISTEN=y
CONFIG_CMD_SEND=y
CONFIG_CMD_TFTP=y
CONFIG_CMD_AT=y
CONFIG_CMD_ADC=y
CONFIG_CMD_I2C=y
CONFIG_CMD_I2C_BMP180=y
# CONFIG_CMD_I2C_SHT21 is not set
# CONFIG_CMD_I2C_BH1750 is not set
# CONFIG_CMD_I2C_PCF8591 is not set
# CONFIG_CMD_I2C_INA219 is not set
# CONFIG_CMD_I2C_TCS3414CS is not set
# CONFIG_CMD_I2C_MLX90614 is not set
# CONFIG_CMD_I2C_HMC5883L is not set
CONFIG_CMD_DS18B20=y
# CONFIG_CMD_DS18B20_DEBUG is not set
CONFIG_CMD_DHT11=y
# CONFIG_CMD_DHT11_DEBUG is not set

#
# Environment Settings
#
CONFIG_ENV_OFFSET=0x7f000
CONFIG_ENV_LEN=0x1000
CONFIG_ADDR_BLOBSETTING1=0x7d000
CONFIG_ADDR_BLOBSETTING2=0x7e000
# CONFIG_ENV_NOWRITE is not set

#
# Deployment settings
#
# CONFIG_DEPLOY_ROOT is not set

#
# Pick the deployment methods you need below
#
CONFIG_DEPLOY_ESPTOOL=y
CONFIG_DEPLOY_ESPTOOL_BIN="esptool.py"
CONFIG_DEPLOY_ESPTOOL_PORT="/dev/ttyUSB0"
CONFIG_DEPLOY_ESPTOOL_ADDRESS="0x00000"
# CONFIG_DEPLOY_ESPTOOL_PL2303 is not set
# CONFIG_DEPLOY_ESPTOOL_MINICOM is not set
# CONFIG_DEPLOY_RF24TOOL is not set
# CONFIG_DEPLOY_ZMODEM is not set

#
# Custom deployment methods
#

#
# Build System Configuration
#
CONFIG_MAKE_DEFTARGET="build"
CONFIG_DEPLOY_DEFTARGET="esptool"
# CONFIG_NOCOLOR is not set
# CONFIG_THREADED is not set
CONFIG_IMAGE_FILENAME="antares"
CONFIG_IMAGE_DIR="images"
# CONFIG_NEED_GENERATE is not set
# CONFIG_BUILD_VERBOSE is not set

#
# Version Information
#

#
# +++ For reference only +++
#

#
# All values below are to be filled by Make scripts
#

#
# Use make set_version to change
#
CONFIG_VERSION_MAJOR="0"
CONFIG_VERSION_MINOR="2-rc1"
CONFIG_VERSION_CODENAME="Insane Mushroom"
CONFIG_VERSION_STRING="0.2-rc1, Insane Mushroom"
CONFIG_VERSION_GIT="2f44651a78db7868efcc53c45f164060cf5ab122"
```

nekromant, 2015-03-19:
Thanks for reporting, I'll give a it a spin in the evening.

d-a-v, 2015-03-19:
Hi

ets_uart_printf is in antares' missing-includes:

$ grep -r ets_uart_printf antares/src/arch/esp8266/include-sdk/
antares/src/arch/esp8266/include-sdk/espmissingincludes.h:int ets_uart_printf(const char *fmt, ...)                         __attribute__((format(printf, 1, 2)));

do you have it ?

maybe you could try and recompile after
	mv antares antares.old

david

dvv, 2015-03-19:
My! Thanks a bunch! Works.

closed 2015-03-19 by dvv