nekromant/esp8266-frankenstein #40: Can't compile i2c_bmp180.c

ystrem, 2015-03-19:
```
ystrem@ubuntu:~/esp8266-frankenstein$ make
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/edeploy.mk
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/deploy.mk
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/arch.mk
  [INFO]      Config changed, running silentoldconfig
  [CHECK]     bash
  [CHECK]     sed
  [CHECK]     awk
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/arch.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/all-arch.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/deploy.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/all-deploy.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/edeploy.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/all-edeploy.kcnf
  [VERSION]   0.2-rc1, Insane Mushroom; git commit: 64304081377ff8ef93034af3e7bc2814e1fe588c  
  [INFO]      Building application code
  [CC]        main.o
  [CC]        stdio.o
  [CC]        helpers.o
  [CC]        microrl.o
  [CC]        console.o
  [CC]        hostname.o
  [CC]        env.o
  [CC]        cb.o
  [CC]        cbtools.o
  [CC]        tcpservice.o
  [CC]        svc_telnet.o
  [CC]        cmd_common.o
  [CC]        cmd_env.o
  [CC]        uart.o
  [CC]        cmd_iw.o
  [CC]        cmd_uart.o
  [CC]        cmd_ifconfig.o
  [CC]        cmd_gpio.o
  [CC]        cmd_flash.o
  [CC]        cmd_listen.o
  [CC]        cmd_send.o
  [CC]        cmd_ds18b20.o
  [CC]        cmd_dht11.o
  [CC]        cmd_tftp.o
  [CC]        tftp.o
  [CC]        cmd_at.o
  [CC]        cmd_adc.o
  [CC]        cmd_i2c.o
  [CC]        pin_map.o
  [CC]        i2c_master.o
  [CC]        i2c_bmp180.o
In file included from /home/ystrem/esp8266-frankenstein/src/i2c_bmp180.c:3:0:
/home/ystrem/esp8266-frankenstein/include/console.h:55:26: error: unknown type name 'size_t'
 int vsnprintf(char *str, size_t size, const char *format, va_list ap);
                          ^
make[1]: *** [i2c_bmp180.o] Error 1
make: *** [builtin] Error 2
```

d-a-v, 2015-03-19:
I just made a pull request for that
https://github.com/nekromant/esp8266-frankenstein/pull/42

in the mean time, you can add #include <c-types.h>
https://github.com/d-a-v/esp8266-frankenstein/commit/edd3faf42ed57bc22cecff65d5722a09cc82e8a2

ystrem, 2015-03-19:
Yep, I can compile it now. :]

ystrem, 2015-03-19:
```
ystrem@ubuntu:~/esp8266-frankenstein$ make
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/deploy.mk
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/arch.mk
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/edeploy.mk
  [INFO]      Config changed, running silentoldconfig
  [CHECK]     bash
  [CHECK]     sed
  [CHECK]     awk
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/arch.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/all-arch.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/deploy.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/all-deploy.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/edeploy.kcnf
  [GEN]       /home/ystrem/esp8266-frankenstein/tmp/all-edeploy.kcnf
  [VERSION]   0.2-rc1, Insane Mushroom; git commit: 64304081377ff8ef93034af3e7bc2814e1fe588c  
*
* Restart config...
*
*
* HeatShrink
*
HeatShrink (LIB_HEATSHRINK) [N/y] (NEW) y
  Encoder (LIB_HEATSHRINK_ENCODE) [N/y] (NEW) y
  Decoder (LIB_HEATSHRINK_DECODE) [N/y] (NEW) y
  Use dynamic memory allocation alloc (LIB_HEATSHRINK_DYNAMIC_ALLOC) [N/y] (NEW) y
  Use index during compression (LIB_HEATSHRINK_USE_INDEX) [N/y] (NEW) y
*
* Modules to build
*
Wireless commands (CMD_IW) [Y/n] y
UART commands (CMD_UART) [N/y] (NEW) y
ifconfig/apconfig (CMD_IF) [Y/n] y
GPIO commands (CMD_GPIO) [Y/n] y
FLASH commands (CMD_FLASH) [Y/n] y
Listen on a port (CMD_LISTEN) [Y/n] y
Send data over TCP (CMD_SEND) [Y/n] y
TFTP firmware update support (CMD_TFTP) [Y/n] y
AT to say OK (CMD_AT) [Y/n] y
System ADC support (CMD_ADC) [Y/n] y
*
* I2C support (experimental)
*
I2C support (experimental) (CMD_I2C) [N/y] (NEW) y
  I2C bmp180 pressure sensor (CMD_I2C_BMP180) [N/y] (NEW) y
  I2C sht21 humidity sensor (CMD_I2C_SHT21) [N/y] (NEW) y
  I2C bh1750 light sensor (CMD_I2C_BH1750) [N/y] (NEW) y
  I2C pcf8591 8-bit ADC DAC (CMD_I2C_PCF8591) [N/y] (NEW) y
  I2C ina219 current sensor (CMD_I2C_INA219) [N/y] (NEW) y
  I2C tcs3414cs color sensor (Needs testing) (CMD_I2C_TCS3414CS) [N/y] (NEW) y
  I2C mlx90614 temperature sensor (Needs testing) (CMD_I2C_MLX90614) [N/y] (NEW) y
  I2C hmc5883l 3-Axis Digital Compass (Needs testing) (CMD_I2C_HMC5883L) [N/y] (NEW) y
*
* DHT11 Thermal & Humidity Sensor support (Needs testing)
*
DHT11 Thermal & Humidity Sensor support (Needs testing) (CMD_DHT11) [N/y] (NEW) y
  DHT11 Debugging (CMD_DHT11_DEBUG) [N/y/?] (NEW) y
#
# configuration written to .config
#
  [INFO]      Building application code
  [CC]        main.o
  [CC]        stdio.o
  [CC]        helpers.o
  [CC]        microrl.o
  [CC]        console.o
  [CC]        hostname.o
  [CC]        env.o
  [CC]        cb.o
  [CC]        cbtools.o
[... 109 lines cut ...]
  [IROMIFY]   Moving application code to IROM
  [LD]        images/antares.elf
/home/ystrem/esp8266-frankenstein/build/app/i2c_hmc5883l.o: In function `HMC5883_ReadDegrees':
i2c_hmc5883l.c:(.irom0.text+0x114): undefined reference to `i2c_master_readRegister16'
i2c_hmc5883l.c:(.irom0.text+0x12a): undefined reference to `i2c_master_readRegister16'
i2c_hmc5883l.c:(.irom0.text+0x14a): undefined reference to `i2c_master_readRegister16'
i2c_hmc5883l.c:(.irom0.text+0x166): undefined reference to `i2c_master_readRegister16'
/home/ystrem/esp8266-frankenstein/build/app/i2c_hmc5883l.o: In function `HMC5883_Read':
i2c_hmc5883l.c:(.irom0.text+0x1b0): undefined reference to `i2c_master_writeRegister'
i2c_hmc5883l.c:(.irom0.text+0x1b4): undefined reference to `i2c_master_readRegister8'
i2c_hmc5883l.c:(.irom0.text+0x214): undefined reference to `i2c_master_writeRegister'
i2c_hmc5883l.c:(.irom0.text+0x227): undefined reference to `i2c_master_readRegister8'
/home/ystrem/esp8266-frankenstein/build/app/i2c_hmc5883l.o: In function `HMC5883_SetConfig':
i2c_hmc5883l.c:(.irom0.text+0x23e): undefined reference to `i2c_master_writeRegister'
i2c_hmc5883l.c:(.irom0.text+0x24b): undefined reference to `i2c_master_readRegister8'
i2c_hmc5883l.c:(.irom0.text+0x266): undefined reference to `i2c_master_writeRegister'
i2c_hmc5883l.c:(.irom0.text+0x273): undefined reference to `i2c_master_readRegister8'
i2c_hmc5883l.c:(.irom0.text+0x28c): undefined reference to `i2c_master_writeRegister'
i2c_hmc5883l.c:(.irom0.text+0x2b6): undefined reference to `i2c_master_readRegister8'
i2c_hmc5883l.c:(.irom0.text+0x2cb): undefined reference to `i2c_master_readRegister8'
i2c_hmc5883l.c:(.irom0.text+0x2da): undefined reference to `i2c_master_readRegister8'
collect2: error: ld returned 1 exit status
make: *** [images/antares.elf] Error 1

```

darkbyte-ru, 2015-03-23:
I create pull request to fix "undefined reference" for i2c functions, but for now you can disable HMC5883 in config and it must be compiled ok.

nekromant, 2015-03-23:
@darkbyte-ru You're likely running out of ram. Check if disabling a few other i2c modules and enabling HMC5883 will suddenly make things work.

darkbyte-ru, 2015-03-23:
@nekromant, hmm, it works. I disable all modules except one i2c for hmc.

nekromant, 2015-03-23:
@darkbyte-ru Well, than we've hit the RAM limit again. 
See https://github.com/nekromant/antares/blob/experimental/src/arch/esp8266/arch.mk. 
The most straightforward solution would be increasing FW_FILE_2 addr in both arch.mk and ld file. We can't put the maximum addresses there or we'll find ourselves back in the situation where firmware code starts at 256KiB leaving little possibilities for OTA updates.
However, the cleaner approach would be calculating the RAM requirements dynamically before actually linking and generating a proper ld file on the run. 
That would basically require us to run xtensa-lx106-elf-size on every .o file, parse the output to get the estimation, round up to the next 4K and use it as FW_FILE_2

darkbyte-ru, 2015-03-24:
@nekromant, what is the problem to write received by tftp data directly to flash? Only one time a got unexpected reset of esp while downloading new firmware via tftp.

nekromant, 2015-03-24:
@darkbyte-ru We'll crash this way. When you are downloading something lwip and the blobs are at work. Sooner or later you'll just overwrite the functions being called from flash and crash. 
the code that does the update MUST reside in RAM. But there's not enough RAM for all the blobs and lwip parts required to work. So the only possible way for this to work - buffer the whole firmware somewhere (we only have flash for that), and then overwrite the previous code by it. The code required to do copying is small and fits in ram. See commit_handler in cmd_tftp.c

nekromant, 2015-03-24:
Closing the issue, since BMP180 compiles in current master

closed 2015-03-24 by nekromant