nekromant/esp8266-frankenstein #54: Telnet need authorisation

nekromant, 2015-03-24:
Otherwise we'll forget about it.

d-a-v, 2015-03-24:
On mar., mars 24, 2015 at 08:09:01 -0700, Andrew wrote:
> Otherwise we'll forget about it. 

Do you mean a password ?

I've been thinking of a no-auth telnet but auth required for
sensible commands through telnet so anyone could read sensors
but not reset nor tftp update nor show env (showing passwords), ...
unless authorized through a sort of sudo command.

nekromant, 2015-03-24:
That mode would require a lot of hackery as I see it. e.g. add permission section for every command. Just add a plain login/password prompt that can be turned off. Automated scripts will have no problem to supply that.

darkbyte-ru, 2015-03-24:
For single user mode only password check will be enough. If password was not set in env, so allow auth to everyone.

nekromant, 2015-03-24:
@darkbyte-ru Sound's like it should be enough.

nekromant, 2018-01-13:
Implemented in current master, closing this issue.

closed 2018-01-13 by nekromant