nekromant/esp8266-frankenstein #56: lwip: Update to upstream.

nekromant, 2015-03-24:
There's already a tree with upstream lwip for esp8266. 
The job at hand is: 
* Analyze the changes to lwip (except for ICACHE_FLASH_ATTR)
* Remove lwip from frankenstein and move it to antares. Antares tree already has all the infrastructure
needed to configure lwip via menuconfig including all the parameters

d-a-v, 2015-04-16:
Hi,
I'm looking for testers on d-a-v/esp8266-frankenstein.git branch lwipupdate.

d-a-v, 2015-04-26:
Done :)
But LWIP git repo is not yet taggued nor versionned. Today's version in frankenstein is 20-apr-2015.
Scripts are provided to make proper diffs against matching LWIP commits. Upgrading must be done with care because of data structures frozen by esp-blobs.
Interesting configurable parameters are TCP_MSS and TCP_WND for a trade-off between speed and ram. IPV6 seems impossible to me since structures would not match blob's ones
The last patch about fixing timers proves itself right : both lwip version (espressif one, and lwip-git) have been running for more than 24hours, tansferring about 20GBytes with no freeze nor leaks and all timeout (poll callback, tcp retransmit timeouts and others) still running good.
Testers are welcome !

nekromant, 2015-04-26:
Yep, closing as fixed. Thanks a lot for your effort making this happen!

closed 2015-04-26 by nekromant