nekromant/esp8266-frankenstein #62: New software Build Fails

Jeshwanth, 2015-03-25:
Hi,

Today I cloned new software and following proper methods to compile, but getting following Error. Any idea ?

```
[jeshwanth@archjeshwanth esp8266-frankenstein]$ make build
  [INFO]      Config changed, running silentoldconfig
  [INFO]      Building application code
  [CC]        main.o
  [CC]        stdio.o
  [CC]        helpers.o
  [CC]        flash_end_marker.o
  [CC]        microrl.o
  [CC]        console.o
  [CC]        hostname.o
  [CC]        env.o
  [CC]        cmd_common.o
  [CC]        cmd_env.o
  [CC]        uart.o
  [CC]        pin_map.o
  [GEN]       built-in.o
  [INFO]      Building antares library code and startup
  [CC]        newlib-dummies.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [IROMIFY]   Moving application code to IROM
  [LD]        images/antares.elf
/home/jeshwanth/mywork/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: images/antares.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
/home/jeshwanth/mywork/esp8266/temp/esp8266-frankenstein/antares/toolchains/gcc.mk:158: recipe for target 'images/antares.elf' failed
make: *** [images/antares.elf] Error 1

```

nekromant, 2015-03-25:
Most likely you've turned off the option 'Use the iROM version of libc' or you're using outdated antares tree with a bug in arch.mk. 
In the latter case - pull the latest git of frankenstein and antares. (I've just added a convenience 'make update' command that fetches updates for both antares and frankenstein and rebases your changes (if any) on top of them)
Afterwards, run 
```
make defconfig
make
```

closed 2015-03-25 by nekromant

susisstrolch, 2015-03-25:
suboptimal...

git checkout master
rm -rf ./antares    # as suggested by mrproper
rm .config
git fetch upstream
git merge upstream/master

strolch@strolch 
/home/strolch/Development/Espressif/share/Repos/frankenstein $ ls
binary  configs  deploy  include  kcnf  Makefile  README.env 
README.hacking  README.md  src
strolch@strolch 
/home/strolch/Development/Espressif/share/Repos/frankenstein $ make 
defconfig
Reverting to default config
strolch@strolch 
/home/strolch/Development/Espressif/share/Repos/frankenstein $ make
ln -sf 
/home/strolch/Development/Espressif/share/Repos/frankenstein/antares 
./antares
Using global antares installation: 
/home/strolch/Development/Espressif/share/Repos/frankenstein/antares
Symlinking done, please re-run make
make: stat: antares: Too many levels of symbolic links

nekromant, 2015-03-25:
@susisstrolch Ooops, forgot to push one last change.

yesco, 2015-09-16:
Can't event get it to compile, even after struggeling to install antiles kconfig etc (terrible process), make was popping up antiles the whole time and wanted me to set targets but there is no esp option anywhere under any CPU... ran make defconfig, then make and then got into this loop...

make
  [INFO]      Config changed, running silentoldconfig
mkdir: cannot create directory `include/config': File exists
/home/knoppix/GIT/esp8266-frankenstein/antares/kconfig/kconfig.mk:75: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 1
/home/knoppix/GIT/esp8266-frankenstein/antares/kconfig/kconfig.mk:70: recipe for target '/home/knoppix/GIT/esp8266-frankenstein/tmp/.configured' failed
make: *** [/home/knoppix/GIT/esp8266-frankenstein/tmp/.configured] Error 2
knoppix@Microknoppix:~/GIT/esp8266-frankenstein$ make clean
  [CLEAN]     cleaning up build directory and temporary files
knoppix@Microknoppix:~/GIT/esp8266-frankenstein$ make
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/edeploy.mk
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/deploy.mk
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/arch.mk
  [INFO]      Config changed, running silentoldconfig
  [CHECK]     bash
  [CHECK]     sed
  [CHECK]     awk
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/arch.kcnf
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/all-arch.kcnf
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/deploy.kcnf
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/all-deploy.kcnf
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/edeploy.kcnf
  [GEN]       /home/knoppix/GIT/esp8266-frankenstein/tmp/all-edeploy.kcnf
  [VERSION]   0.2-rc1, Insane Mushroom; git commit: 7d7de5ffeafb228d8688cc17b4b49063c2550648  
mkdir: cannot create directory `include/config': File exists
/home/knoppix/GIT/esp8266-frankenstein/antares/kconfig/kconfig.mk:75: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 1
/home/knoppix/GIT/esp8266-frankenstein/antares/kconfig/kconfig.mk:70: recipe for target '/home/knoppix/GIT/esp8266-frankenstein/tmp/.configured' failed
make: *** [/home/knoppix/GIT/esp8266-frankenstein/tmp/.configured] Error 2

nekromant, 2015-09-18:
@yesco You need experimental branch of antares to compile frankenstein. esp8266 is not yet in stable branch. I plan on mergining experimental into main somewhere in october, since quite a few bugs need fixing.