nekromant/esp8266-frankenstein #64: make mrproper results in "Error 1"

susisstrolch, 2015-03-25:
it seems that "make mrproper" (on current master 82ae8af) doesn't clean inside ./antares:
>strolch@strolch /home/strolch/Development/Espressif/share/Repos/frankenstein $ make mrproper
  [GEN]       /home/strolch/Development/Espressif/share/Repos/frankenstein/tmp/edeploy.mk
  [GEN]       /home/strolch/Development/Espressif/share/Repos/frankenstein/tmp/deploy.mk
  [GEN]       /home/strolch/Development/Espressif/share/Repos/frankenstein/tmp/arch.mk
  [CLEAN]     cleaning up build directory and temporary files
  [MRPROPER]  purging everything
rm: cannot remove ‘/home/strolch/Development/Espressif/share/Repos/frankenstein/antares’: Is a directory
make: *** [mrproper] Error 1

nekromant, 2015-03-25:
Is antares in your project root a directory or a symlink to a system-wide installation?

susisstrolch, 2015-03-25:
It's a regular directory - created by the initial "make" of frankenstein.

Ah - the mrproper also deletes the images/ directory, but it doesn't get 
rebuild by make or make build...

nekromant, 2015-03-25:
Actually, it does. But due to failing to delete antares dir mrproper doesn't remove all the other stuff so next time you run build the initial directory creation magic is not run. 
Since I use antares for a ton of project the usual mode is 'symlink' antares to each project.  
I have committed fixes in both antares tree and frankenstein that should fix this issue for now. Try a clean build.

closed 2015-03-25 by nekromant