nekromant/esp8266-frankenstein #66: Vdd command seems unstable

darkbyte-ru, 2015-03-25:
New command "vdd" return strange values:
```
frankenstein > vdd
VDD3V3 = 3430 mV
```
But esp powered by LiIon battery through AMS1117 regulator, and multimeter show 3.286 V in module VCC pin.

Also, command crash module randomly (after 1-10 runs).

I try system_get_vdd33 function, but it always return 65535.

nekromant, 2015-03-25:
That insect's definitely coming from Espressif's blobbies. The command's mostly undocumented so I'm not sure whether this is raw adc value or actual mV (mV was a blind guess).

darkbyte-ru, 2015-03-25:
Yep, I understand. What about reading vdd for others?

susisstrolch, 2015-03-25:
Maybe I'm still with the old SDK...
How do I update the SDK in antares build system?

darkbyte-ru, 2015-03-25:
@susisstrolch, 
```
cd where/is/src/esp-open-sdk
make clean
git pull
git submodule update
make STANDALONE=y
```

susisstrolch, 2015-03-25:
Here the result (SDK1.0 based):

frankenstein > vdd
VDD3V3 = 98 mV

frankenstein > vdd
VDD3V3 = 98 mV

frankenstein > vdd

  ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

nekromant, 2015-03-25:
@susisstrolch There's something bad and broken about 1.0 SDK, and I can't (yet) figure out what,

susisstrolch, 2015-03-25:
No worry...
just for the files:
running "vdd" via serial console reboots after approx 2-3 tries.
In contrast, running vdd in a telnet session doesn't reboot the esp, but 
shows approx. 84mV...

nekromant, 2015-03-25:
And it gets weirder and weirder. My module shows the same voltage over telnet as it does via serial. Though I did manage to get one crash and reboot while running this command.

nekromant, 2015-03-25:
@susisstrolch Try the current master. Since I can barely reproduce it on my hardware - see if this hack helps.

susisstrolch, 2015-03-25:
no wdt reset after lots of vdd readouts.
Readout toggling between 114mV and 115mV.

Maybe a speciality of my dev board...

nekromant, 2015-03-25:
@susisstrolch Hm, I'm not really sure where this vdd is being sampled from. May be it's some pin that's not wired.

d-a-v, 2015-03-25:
Have you tried TOUT pin and adc command ?
It seems stable and seems to measure mV with no conversion 1.024V max.
The measurement itself is not really stable at 10bits but it is worth trying.

nekromant, 2015-03-26:
@d-a-v They're not wired out on my modules. I only have access to RST, RX, TX, GPIO0, GPIO2

susisstrolch, 2015-03-26:
Short summary from talk with espressif:
As soon as TOUT isn't left open, the "read_vdd33" will not return any 
meaningfull result.
My board has TOUT connected to a LDR/resistor divider, so I'll never see 
the real Vdd.

nekromant, 2015-03-26:
@susisstrolch Thanks! I guess we can close this issue now.

closed 2015-03-26 by nekromant

susisstrolch, 2015-03-26:
send you a pm to your contact address...

darkbyte-ru, 2015-03-27:
That is strange. I test read_vdd33 on esp-01 board and it seems that TOUT not wired to anything.

Also, I discovered, that chip crashes only in AP mode and work ok in APSTA.

susisstrolch, 2015-03-27:
Thats exactly the case where read_vdd33 is supposed to work.
According to Espressif (NDA info), read_vdd33 only works if TOUT is 
floating.
As soon as it's connected (like mine - to an R-LDR divider) it will show 
you complete
different values - probably the voltage itself.
Espressifs explanation is a bit complicate and lengthy - and I still 
don't understand all implications.

d-a-v, 2015-03-27:
update, vdd seems to be working (esp-01 with TOUT soldered out), and valid only if TOUT is floating.
(no more reset)

I am on SDK 1.0.0 synchronized with master (at the time of writing) which was not the case on my previous message above.

-------------------
TOUT floating:
frankenstein > vdd
VDD3V3 = 3413 mV
frankenstein > vdd
VDD3V3 = 3415 mV

-------------------
TOUT to GND:
frankenstein > vdd
VDD3V3 = 15 mV
frankenstein > vdd
VDD3V3 = 16 mV
frankenstein > adc
3
frankenstein > vdd
VDD3V3 = 16 mV
frankenstein > vdd
VDD3V3 = 16 mV

-------------------
TOUT to 3.3V:
frankenstein > vdd
VDD3V3 = 4094 mV
frankenstein > adc
1024

-------------------
TOUT floating:
frankenstein > vdd
VDD3V3 = 3422 mV
frankenstein > adc
1024
frankenstein > adc
1024
frankenstein > adc
1024
frankenstein > adc
0
frankenstein > vdd
VDD3V3 = 3414 mV

nekromant, 2015-03-27:
@d-a-v I'm not sure TOT can handle 3.3v correctly. In your case your adc values are all maxed out. Can you try with something like 250mV 500mV, 1V?

d-a-v, 2015-03-27:
> @d-a-v I'm not sure TOT can handle 3.3v correctly. In your case your adc values are all maxed out. Can you try with something like 250mV 500mV, 1V?

It cannot, you are right.

TOUT seems to give mV (1024 would then be 1.024V although I'm not sure whether it is 1.000V or 1.024V).

It is stuck at 1024 with above voltage (I did try not feed it with more than 3.3v).