nekromant/esp8266-frankenstein #70: Moving logic code into libraries

pastcompute, 2015-04-01:
Hi, this is just an option for discussion, it depends on which way you want to take this.

I have a project I'm basing on frankenstein and one of the things I will be doing is writing a program that schedules a timer task to poll a bunch of sensors. To this end it included copying the bulk of the code out of ds18b20.c to another file and making the entry point extern so I could call it from my code.

Longer term for most of the commands like ds18b20 that do similar things, they could be improved by making them expose an internal API (usually, just a single extern function) that the static method that parses the argv would call, that could then be called by other code to use the same functionality.

The tcp_send command is another similar example

If you like I can submit that change for ds18b20 and dht22 as a proof of concept

If this is not suitable I'm not fussed, I'll just keep a local fork of relevant logic code in my project

susisstrolch, 2015-04-02:
Great idea - I was also thinking about...
Having the lib and the cmd stub makes it much easier to develop own projects!

nekromant, 2015-04-02:
Yep, this is something I wanted to do, prior to moving these to antares itself. 
However, before that I still planned to: 
1. Implement delaylib wrapper for esp8266 (cross-mcu delays, AFAIR dht22 uses delays for timing. Using antares' delaylib will instantly make it work on all other architectures)
2. Implement a generic i2c subsystem for antares and rebase all i2c sensors on-top of it. 
...
PROFIT

darkbyte-ru, 2015-04-02:
I can split i2c modules to two parts: library for antares and command for frankenstein, but I do not know how to do it better way.

Tons of i2c_* files seems terrible to me. I try to combine all i2c libs in one file and leave separated cmd files, and it's look much more terrible :D

Maybe try combine all cmd_i2c in one file (with ifdef's for every driver) and leave separated i2c drivers?

nekromant, 2015-04-02:
@darkbyte-ru Don't do it just yet, give me a little time to make an proper i2c subsystem for antares.