nekromant/esp8266-frankenstein #74: Minor problems with build instructions

Juul, 2015-04-13:
It might be good to tell people to run:
```
make defconfig
```
before running make menuconfig, to start them out with sane defaults.

Then, after make menuconfig they need to run:
```
mkdir images
```
before running:
```
make build
```

ystrem, 2015-04-17:
Thank you !!! :D

digger3d, 2015-04-23:
Unfortunately after I followed the procedure of installing everything I am still getting errors. Please help me to pass this step. Here is a log:
anton@Satellite-A105:~/esp8266-frankenstein$ make defconfig
Reverting to default config

anton@Satellite-A105:~/esp8266-frankenstein$ make build
  [INFO]      Config changed, running silentoldconfig
*
* Restart config...
*
*
* Target platform settings
*
Target architecture
> 1. 8051 (ARCH_8051)
  2. arm (ARCH_ARM)
  3. avr (ARCH_AVR)
  4. gbz80 (ARCH_GBZ80)
  5. mips (ARCH_MIPS)
  6. msp430 (ARCH_MSP430)
  7. native (ARCH_NATIVE)
  8. pic32 (ARCH_PIC32)
choice[1-8]: 1
Pick an MCU family
> 1. STC 8051 family (ARCH_8051_STC) (NEW)
  2. Atmel AT89 family (ARCH_8051_AT89) (NEW)
choice[1-2]: 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 24
Memory model
> 1. Small (MODEL_SMALL) (NEW)
  2. Medium (MODEL_MEDIUM) (NEW)
  3. Large (MODEL_LARGE) (NEW)
  4. Huge (MODEL_HUGE) (NEW)
choice[1-4]: 
XRAM size (bytes) (XRAM_SIZE) [128] (NEW) 
IRAM size (bytes) (IRAM_SIZE) [128] (NEW) 
FLASH size (bytes) (FLASH_SIZE) [4096] (NEW) 
*
* Platform-specific libraries
*
STC IAP/ISP routines (LIB_STC_IAP) [N/y/?] (NEW) 
*
* Toolchain settings
*
Show all toolchains (SHOW_ALL_TOOLCHAINS) [N/y] n
Toolchain selection
> 1. SDCC-based toolchain (TOOLCHAIN_SDCC) (NEW)
choice[1]: 1
Extra SDCC CFLAGS (SDCC_CFLAGS) [] (NEW) 
Extra SDCC ASFLAGS (SDCC_ASFLAGS) [] (NEW) 
Extra SDCC LDFLAGS (SDCC_LDFLAGS) [] (NEW) 
*
* Services to build
*
DHCP Server (SERVICE_DHCPS) [Y/n/?] y
TCP/IP services (SERVICE_TCP) [Y/n] y
  Telnet Service (SERVICE_TELNET) [Y/n] y
  Echo Service (SERVICE_ECHO) [N/y] (NEW) 
*
* stcdude
*
stcdude (DEPLOY_STCDUDE) [N/y] (NEW) 
#
# configuration written to .config
#
  [INFO]      Building application code
  [CC]        main.o
xtensa-lx106-elf-gcc: error: unrecognized command line option '-mmcu='
/home/anton/esp8266-frankenstein/antares/make/Makefile.build:78: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/anton/esp8266-frankenstein/antares/toolchains/gcc.mk:119: recipe for target 'builtin' failed
make: *** [builtin] Error 2

nekromant, 2015-04-23:
Ah, you're running antares mater branch. You'll need check out antares experimental branch. ESP8266 port is just  not very stable. Yet.

closed 2015-04-23 by nekromant

digger3d, 2015-04-23:
Thank you for the fast reply. Ok, Im starting over.

digger3d, 2015-04-23:
1st I got this
anton@Satellite-A105:~/.antares_examples/esp8266-frankenstein$ make
ln -sf ./antares ./antares
Using global antares installation: ./antares
Symlinking done, please re-run make
make: stat: antares: Too many levels of symbolic links

anton@Satellite-A105:~/.antares_examples/esp8266-frankenstein$ make
make: stat: antares/Makefile: Too many levels of symbolic links
make: stat: antares: Too many levels of symbolic links
ln -sf ./antares ./antares
Using global antares installation: ./antares
Symlinking done, please re-run make
make: stat: antares: Too many levels of symbolic links
then I linked ~/.antares into ~/.antares_examples/esp8266-frankenstein as ~/.antares_examples/esp8266-frankenstein/antares
and finally got this error:
anton@Satellite-A105:~/.antares_examples/esp8266-frankenstein$ make defaults
make: *** No rule to make target 'defaults'.  Stop.
anton@Satellite-A105:~/.antares_examples/esp8266-frankenstein$ make defconfig
Reverting to default config
anton@Satellite-A105:~/.antares_examples/esp8266-frankenstein$ make build
  [INFO]      Config changed, running silentoldconfig
*
* Restart config...
*
*
* Target platform settings
*
Target architecture
> 1. 8051 (ARCH_8051)
  2. arm (ARCH_ARM)
  3. avr (ARCH_AVR)
  4. gbz80 (ARCH_GBZ80)
  5. mips (ARCH_MIPS)
  6. msp430 (ARCH_MSP430)
  7. native (ARCH_NATIVE)
  8. pic32 (ARCH_PIC32)
choice[1-8]: 1
Pick an MCU family
> 1. STC 8051 family (ARCH_8051_STC) (NEW)
  2. Atmel AT89 family (ARCH_8051_AT89) (NEW)
choice[1-2]: 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 24
Memory model
> 1. Small (MODEL_SMALL) (NEW)
  2. Medium (MODEL_MEDIUM) (NEW)
  3. Large (MODEL_LARGE) (NEW)
  4. Huge (MODEL_HUGE) (NEW)
choice[1-4]: 
XRAM size (bytes) (XRAM_SIZE) [128] (NEW) 
IRAM size (bytes) (IRAM_SIZE) [128] (NEW) 
FLASH size (bytes) (FLASH_SIZE) [4096] (NEW) 
*
* Platform-specific libraries
*
STC IAP/ISP routines (LIB_STC_IAP) [N/y/?] (NEW) 
*
* Toolchain settings
*
Show all toolchains (SHOW_ALL_TOOLCHAINS) [N/y] n
Toolchain selection
> 1. SDCC-based toolchain (TOOLCHAIN_SDCC) (NEW)
choice[1]: 1
Extra SDCC CFLAGS (SDCC_CFLAGS) [] (NEW) 
Extra SDCC ASFLAGS (SDCC_ASFLAGS) [] (NEW) 
Extra SDCC LDFLAGS (SDCC_LDFLAGS) [] (NEW) 
*
* Services to build
*
DHCP Server (SERVICE_DHCPS) [Y/n/?] y
TCP/IP services (SERVICE_TCP) [Y/n] y
  Telnet Service (SERVICE_TELNET) [Y/n] y
  Echo Service (SERVICE_ECHO) [N/y] (NEW) 
*
* stcdude
*
stcdude (DEPLOY_STCDUDE) [N/y] (NEW) 
#
# configuration written to .config
#
  [INFO]      Building application code
  [CC]        main.o
xtensa-lx106-elf-gcc: error: unrecognized command line option '-mmcu='
/home/anton/.antares_examples/esp8266-frankenstein/antares/make/Makefile.build:78: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/anton/.antares_examples/esp8266-frankenstein/antares/toolchains/gcc.mk:119: recipe for target 'builtin' failed
make: *** [builtin] Error 2
what is wrong?

digger3d, 2015-04-23:
when I try to build from a different location (also had to symlink) I am getting this log:

anton@Satellite-A105:~/esp8266-frankenstein$ make defconfig
Reverting to default config
anton@Satellite-A105:~/esp8266-frankenstein$ make
  [INFO]      Config changed, running silentoldconfig
*
* Restart config...
*
*
* Target platform settings
*
Target architecture
> 1. 8051 (ARCH_8051)
  2. arm (ARCH_ARM)
  3. avr (ARCH_AVR)
  4. gbz80 (ARCH_GBZ80)
  5. mips (ARCH_MIPS)
  6. msp430 (ARCH_MSP430)
  7. native (ARCH_NATIVE)
  8. pic32 (ARCH_PIC32)
choice[1-8]: 1
Pick an MCU family
> 1. STC 8051 family (ARCH_8051_STC) (NEW)
  2. Atmel AT89 family (ARCH_8051_AT89) (NEW)
choice[1-2]: 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 
CPU Frequency in Hz (F_CPU) [] (NEW) 52
Memory model
> 1. Small (MODEL_SMALL) (NEW)
  2. Medium (MODEL_MEDIUM) (NEW)
  3. Large (MODEL_LARGE) (NEW)
  4. Huge (MODEL_HUGE) (NEW)
choice[1-4]: 
XRAM size (bytes) (XRAM_SIZE) [128] (NEW) 
IRAM size (bytes) (IRAM_SIZE) [128] (NEW) 
FLASH size (bytes) (FLASH_SIZE) [4096] (NEW) 
*
* Platform-specific libraries
*
STC IAP/ISP routines (LIB_STC_IAP) [N/y/?] (NEW) 
*
* Toolchain settings
*
Show all toolchains (SHOW_ALL_TOOLCHAINS) [N/y] n
Toolchain selection
> 1. SDCC-based toolchain (TOOLCHAIN_SDCC) (NEW)
choice[1]: 1
Extra SDCC CFLAGS (SDCC_CFLAGS) [] (NEW) 
Extra SDCC ASFLAGS (SDCC_ASFLAGS) [] (NEW) 
Extra SDCC LDFLAGS (SDCC_LDFLAGS) [] (NEW) 
*
* Services to build
*
DHCP Server (SERVICE_DHCPS) [Y/n/?] y
TCP/IP services (SERVICE_TCP) [Y/n] y
  Telnet Service (SERVICE_TELNET) [Y/n] y
  Echo Service (SERVICE_ECHO) [N/y] (NEW) 
*
* stcdude
*
stcdude (DEPLOY_STCDUDE) [N/y] (NEW) 
#
# configuration written to .config
#
  [INFO]      Building application code
  [CC]        main.o
xtensa-lx106-elf-gcc: error: unrecognized command line option '-mmcu='
/home/anton/esp8266-frankenstein/antares/make/Makefile.build:78: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/anton/esp8266-frankenstein/antares/toolchains/gcc.mk:119: recipe for target 'builtin' failed
make: *** [builtin] Error 2

digger3d, 2015-04-23:
what else I've got in my mind:

anton@Satellite-A105:~/esp8266-frankenstein$ make update
git fetch && git rebase -i
Cannot rebase: You have unstaged changes.
Please commit or stash them.
Makefile:47: recipe for target 'update' failed
make: *** [update] Error 1

nekromant, 2015-04-24:
cd antares && git checkout experimental