nekromant/esp8266-frankenstein #77: building

digger3d, 2015-04-22:
anton@Satellite-A105:~/esp8266-frankenstein$ make build
  [INFO]      Building antares library code and startup
  [GEN]       built-in.rel
  [GEN]       built-in.rel
  [GEN]       built-in.rel
  [GEN]       built-in.rel
  [GEN]       built-in.rel
  [GEN]       built-in.rel
  [GEN]       built-in.rel
  [GEN]       built-in.rel
  [INFO]      Building application code
  [CC]        main.rel
/home/anton/esp8266-frankenstein/src/main.c:4:21: fatal error: ets_sys.h: No such file or directory
compilation terminated.
/home/anton/esp8266-frankenstein/antares/make/Makefile.build:78: recipe for target 'main.rel' failed
make[1]: *** [main.rel] Error 1
/home/anton/esp8266-frankenstein/antares/toolchains/sdcc.mk:38: recipe for target 'builtin' failed
make: *** [builtin] Error 2

nekromant, 2015-04-22:
Why are you trying to compile it with sdcc? 
start with make defconfig
and later adjust when needed,

nekromant, 2015-04-22:
See #74

closed 2015-04-22 by nekromant