nekromant/esp8266-frankenstein #79: Building failed

digger3d, 2015-04-24:
After numerous attempts to build all needed components of esp8266-frankenstein, smoking of all the READMEs and docs in antares, esp-open-sdk, kconfig-fronends and esp8266-frankenstein itself what do I do now after finally getting to this point:

anton@u64:~/esp8266-frankenstein$ mkdir images
anton@u64:~/esp8266-frankenstein$ make defconfig
  [GEN]       /home/anton/esp8266-frankenstein/tmp/edeploy.mk
  [GEN]       /home/anton/esp8266-frankenstein/tmp/deploy.mk
  [GEN]       /home/anton/esp8266-frankenstein/tmp/arch.mk
Reverting to default config
anton@u64:~/esp8266-frankenstein$ make
  [INFO]      Config changed, running silentoldconfig
  [CHECK]     bash
  [CHECK]     sed
  [CHECK]     awk
  [GEN]       /home/anton/esp8266-frankenstein/tmp/arch.kcnf
  [GEN]       /home/anton/esp8266-frankenstein/tmp/all-arch.kcnf
  [GEN]       /home/anton/esp8266-frankenstein/tmp/deploy.kcnf
  [GEN]       /home/anton/esp8266-frankenstein/tmp/all-deploy.kcnf
  [GEN]       /home/anton/esp8266-frankenstein/tmp/edeploy.kcnf
  [GEN]       /home/anton/esp8266-frankenstein/tmp/all-edeploy.kcnf
  [VERSION]   0.2-rc1, Insane Mushroom; git commit: 7133d1f6f32c5cb0dc754216d0188ca2542b0cdb
*
* Restart config...
*
*
* Services to build
*
DHCP Server (SERVICE_DHCPS) [Y/n/?] y
TCP/IP services (SERVICE_TCP) [Y/n] y
  Telnet Service (SERVICE_TELNET) [Y/n] y
  Echo Service (SERVICE_ECHO) [N/y] (NEW)
#
# configuration written to .config
#
  [INFO]      Building application code
  [CC]        main.o
  [CC]        stdio.o
  [CC]        helpers.o
  [CC]        flash_end_marker.o
  [CC]        microrl.o
  [CC]        console.o
  [CC]        hostname.o
  [CC]        env.o
  [CC]        cbuf.o
  [CC]        cbuftools.o
  [CC]        tcpservice.o
  [CC]        svc_telnet.o
  [CC]        cmd_common.o
/home/anton/esp8266-frankenstein/src/cmd_common.c: In function 'do_vdd':
/home/anton/esp8266-frankenstein/src/cmd_common.c:40:2: warning: implicit declaration of function 'ets_intr_lock' [-Wimplicit-function-declaration]
  os_intr_lock();
  ^
/home/anton/esp8266-frankenstein/src/cmd_common.c:42:2: warning: implicit declaration of function 'ets_intr_unlock' [-Wimplicit-function-declaration]
  os_intr_unlock();
  ^
  [CC]        cmd_env.o
  [CC]        uart.o
  [CC]        pin_map.o
  [CC]        iwconnect.o
  [CC]        cmd_iw.o
  [CC]        cmd_uart.o
  [CC]        cmd_ifconfig.o
  [CC]        cmd_gpio.o
  [CC]        cmd_flash.o
  [CC]        cmd_listen.o
  [CC]        cmd_send.o
  [CC]        cmd_ds18b20.o
  [CC]        cmd_dht22.o
  [CC]        cmd_tftp.o
  [CC]        tftp.o
  [CC]        cmd_at.o
  [CC]        cmd_adc.o
  [CC]        cmd_nslookup.o
  [CC]        cmd_i2c.o
  [CC]        i2c_master.o
  [CC]        i2c_bmp180.o
  [CC]        i2c_ina219.o
  [CC]        i2c_sht21.o
  [CC]        i2c_bh1750.o
  [CC]        i2c_pcf8591.o
  [CC]        i2c_hmc5883l.o
  [CC]        i2c_tcs3414cs.o
  [CC]        i2c_mlx90614.o
  [GEN]       built-in.o
  [CC]        api_lib.o
  [CC]        api_msg.o
  [CC]        err.o
  [CC]        netbuf.o
  [CC]        netdb.o
  [CC]        netifapi.o
  [CC]        sockets.o
  [CC]        tcpip.o
  [GEN]       built-in.o
  [CC]        dhcpserver.o
  [CC]        espconn.o
  [CC]        espconn_tcp.o
  [CC]        espconn_udp.o
In file included from /home/anton/esp8266-frankenstein/antares/src/arch/esp8266/include-sdk/user_interface.h:11:0,
                 from /home/anton/esp8266-frankenstein/src/lwip/app/espconn_udp.c:15:
/home/anton/esp8266-frankenstein/src/lwip/app/espconn_udp.c: In function 'espconn_udp_recv':
/home/anton/esp8266-frankenstein/src/lwip/app/espconn_udp.c:167:31: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   if (!ip_addr_netcmp((struct ip_addr *)precv->pespconn->proto.udp->remote_ip, &ipconfig.ip, &ipconfig.netmask)) {
                               ^
/home/anton/esp8266-frankenstein/include/lwip-esp8266/lwip/ip_addr.h:194:47: note: in definition of macro 'ip_addr_netcmp'
 #define ip_addr_netcmp(addr1, addr2, mask) (((addr1)->addr & \
                                               ^
  [GEN]       built-in.o
  [CC]        def.o
  [CC]        dhcp.o
  [CC]        dns.o
  [CC]        init.o
  [CC]        mem.o
  [CC]        memp.o
  [CC]        netif.o
  [CC]        pbuf.o
  [CC]        raw.o
  [CC]        stats.o
  [CC]        sys_arch.o
  [CC]        sys.o
  [CC]        tcp.o
  [CC]        tcp_in.o
  [CC]        tcp_out.o
  [CC]        timers.o
  [CC]        udp.o
  [CC]        autoip.o
  [CC]        icmp.o
  [CC]        igmp.o
  [CC]        inet.o
  [CC]        inet_chksum.o
  [CC]        ip_addr.o
  [CC]        ip.o
  [CC]        ip_frag.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [CC]        etharp.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [INFO]      Building antares library code and startup
  [CC]        newlib-dummies.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [GEN]       built-in.o
  [IROMIFY]   Moving application code to IROM
  [LD]        images/antares.elf
/home/anton/esp8266-frankenstein/build/app/cmd_at.o:(.irom0.text+0x4): undefined reference to `system_get_sdk_version'
/home/anton/esp8266-frankenstein/build/app/cmd_at.o: In function `do_at':
cmd_at.c:(.irom0.text+0x28): undefined reference to `system_get_sdk_version'
/home/anton/esp8266-frankenstein/build/app/cmd_adc.o:(.irom0.text+0x4): undefined reference to `system_adc_read'
/home/anton/esp8266-frankenstein/build/app/cmd_adc.o: In function `do_adc':
cmd_adc.c:(.irom0.text+0x14): undefined reference to `system_adc_read'
/home/anton/esp8266-frankenstein/build/app/lwip/app/dhcpserver.o: In function `handle_dhcp':
dhcpserver.c:(.irom0.text+0x710): undefined reference to `os_printf_plus'
dhcpserver.c:(.irom0.text+0x747): undefined reference to `os_printf_plus'
/home/anton/esp8266-frankenstein/build/app/lwip/app/espconn.o: In function `espconn_accept':
espconn.c:(.irom0.text+0x502): undefined reference to `os_printf_plus'
/home/anton/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o: In function `espconn_server_recv':
espconn_tcp.c:(.irom0.text+0x9b6): undefined reference to `os_printf_plus'
/home/anton/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o: In function `espconn_tcp_sent':
espconn_tcp.c:(.irom0.text+0x173): undefined reference to `os_printf_plus'
/home/anton/esp8266-frankenstein/build/app/lwip/app/espconn_tcp.o:espconn_tcp.c:(.irom0.text+0x196): more undefined references to `os_printf_plus' follow
collect2: error: ld returned 1 exit status
make: *** [images/antares.elf] Error 1

nekromant, 2015-04-24:
Looks like you have outddated esp-open-sdk. system_get_sdk_version was not around until something like 0.9.6 or 1.0. Use The latest esp-open-sdk, and build it with STANDALONE option.

closed 2015-04-24 by nekromant

digger3d, 2015-04-24:
I did install version 0.9.2 because you wrote this: 
"For the record: frankenstein is based on 0.9.2 SDK. 0.9.3 will likely break some things. 
I plan to move frankenstein to 0.9.3 as soon an pfalcon bumps esp-open-sdk. "
here:
https://github.com/nekromant/esp8266-frankenstein/blob/master/README.hacking

nekromant, 2015-04-24:
Ouch, gitta update the README, thanks for noticing. Bump to 0.9.3 broke quite some things, so I had it to put into the README util I could finally fix it. And o it stayed.

digger3d, 2015-04-24:
Thank you for help, I finally compiled the firmware! Hurrah!!!

digger3d, 2015-04-24:
I also can write a better readme, which will help avoind endless reinstallations