nekromant/esp8266-frankenstein #82: Build against SDK 1.0.1b2 is broken

nekromant, 2015-04-25:
Looks like we have a problem with 1.0.1b2 SDK, something's changed and it can't ever connect to a network. Now trying to see where that insect came from

d-a-v, 2015-04-30:
I did not try, but I came into this: https://github.com/themadinventor/esptool/issues/33
Could it be related ?

nekromant, 2015-04-30:
Unlikely. My guess is that something's changed in headers yet again, but I may be wrong. 
The firmware is working, I can access commandline and even scan for networks, but whenever I try to connect the console becomes laggy and occasionally ends up in an exception and a reboot. I could never connect to a network on 1.0.1b2, while 1.0.1b1 worked awesomely.

susisstrolch, 2015-05-02:
Another flaw - not really the reason for that problem - is the default 
base address for environment.
According to SDK doc, the last 16k are used by the system.
So, the default should be set to 0x7B000 instead 0x7C000.

nekromant, 2017-12-14:
Now fixed, and builds against 2.0

closed 2017-12-14 by nekromant