nekromant/esp8266-frankenstein #93: make error

GTP95, 2018-02-18:
Hello,
I'm really interested in this firmware for the esp8266 and I'd love to experiment with it but I ran into a problem: after installing all required dependencies, creating the "build" folder as suggested and running inside it "cmake .." and "make menuconfig" to tweak the build configuration, running "make" throws the following error:

> $ make
[  0%] Built target frankenstein
[  0%] Built target iromify-frankenstein
make[2]: *** No rule to make target 'CMakeFiles/frank.dir/build'.  Stop.
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/frank.dir/all' failed
make[1]: *** [CMakeFiles/frank.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

What am I doing wrong? 
I also think that it would be a good idea to provide a precompiled rom ready to flash, so users that don't need to change the default config could spare a lot of time instead having to build from source.
Thanks for your help.

nekromant, 2018-02-19:
Please post your cmake log. Have you done git submodule update --init?

You can grab binary builds from Jenkins, see badge in the readme. Sorry for short response - writing this from my mobile phone.

GTP95, 2018-02-20:
Thanks for your quick answer. I've skipped the git submodule update --init, so I did it and then did again cmake, make menuconfig and make. This time I get a different error:

> $ make
[  0%] Generating include/generated/autoconf.h
Scanning dependencies of target frankenstein
[  0%] Building C object CMakeFiles/frankenstein.dir/src/user_rf_cal_sector_set.c.obj
/home/vagrant/esp8266-frankenstein/src/user_rf_cal_sector_set.c:1:21: fatal error: c_types.h: No such file or directory
 #include <c_types.h>
                     ^
compilation terminated.
CMakeFiles/frankenstein.dir/build.make:67: recipe for target 'CMakeFiles/frankenstein.dir/src/user_rf_cal_sector_set.c.obj' failed
make[2]: *** [CMakeFiles/frankenstein.dir/src/user_rf_cal_sector_set.c.obj] Error 1
CMakeFiles/Makefile2:312: recipe for target 'CMakeFiles/frankenstein.dir/all' failed
make[1]: *** [CMakeFiles/frankenstein.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Judging from the CMake log seems a problem with xtensa, if you're curious I will provide you the CMake logs, but I think I'll try first the precompiled binaries and I'll try to recompile form source only if I have to do some modifications.
Thanks again

nekromant, 2018-02-21:
Looks like your have a buggy toolchain that is missing headers. You have to use esp-open-sdk to build frankenstein. I can post my toolchain if you want, or you can build it yourself.

nekromant, 2018-02-21:
You can use this one: https://cloud.ncrmnt.org/index.php/s/jYDxzw52g95cLKP

closed 2018-02-21 by nekromant