nekromant/esp8266-frankenstein #94: filesystem.espfs

GTP95, 2018-02-20:
Just two more questions: at which address should I flash the filesystem.espfs file? Do I flash it the same way as binary files using the write_flash option of esptool or should I use write_mem instead?
Sorry but I don't know esptool well and it's documentation doesn't explain the difference between write_flash and write_mem.

nekromant, 2018-02-21:
filesystem.espfs stores all the static pages served via web. It is flashed in the next block after the firmware, the actual offset is calculated dynamically. Right now you can download the filesystem file via tftp. There's also a webdl command for that, but it's currently crashing. 

You can find out the right address from the fk boot log and use write_flash afterwards.

GTP95, 2018-02-25:
I'm still facing problems with flashing. I can't flash it on an ESP-07, it fails with a "waiting for packet header" error at 93% or 95% depending most of the times on the baud rate I use, but this one doesn't really count since maybe it's memory is already damaged (I can't run a simple sketch, but Arduino IDE is still able to flash it). But what is strange is that I have a nodemcu v3 board (or similar clone with an ESP-12N mounted on it, not sure it's original) that I can flash fine, but right after flashing it I get the following serial output endlessly repeated:

> Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000

and after a reset nothing is printed, just the usual strange output you get during reset. I'm using the following command to flash it with the following output:

> $ esptool.py -p /dev/ttyUSB0 write_flash 0x00000 frank-0x00000.bin 0x10000 frank-0x10000.bin 
esptool.py v1.3
Connecting....
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0040
Wrote 49152 bytes at 0x0 in 4.3 seconds (92.0 kbit/s)...
Wrote 307200 bytes at 0x10000 in 26.6 seconds (92.3 kbit/s)...
Leaving...

I'm doing something wrong? Is there anything else I could try?
Thanks for your support and have a nice day

GTP95, 2018-02-25:
UPDATE: seems that the issue is solved (on the nodemcu) by specifying the "dio" flashing mode, now after flashing I get the following output:

> load 0x40100000, len 28448, room

What should I do now?

nekromant, 2018-02-26:
If you are stuck in a bootloop most likely you are missing esp-init-data.bin. See nodemcu docs, where they describe flashing one. Try to fix that one first.

GTP95, 2018-06-12:
Hello again, I just tried to flash a brand new esp8266 using nodemcu flasher (https://github.com/nodemcu/nodemcu-flasher) (esptool was failing as previously described). The tool flashed it without trowing errors (but was really quick) but once rebooted I got seemingly random characters. I tried to set my terminal to 9600 and 115200 baud but didn't got any intelligible output, which is the default baud rate of your firmware?

nekromant, 2018-06-13:
It should be 115200! ROM loader spews out some debugging at weird 78kbps or so, so a little bit of gibberish is expected with esp8266 on any firmware.

nekromant, 2018-06-14:
I'm using a variety of esp8266 modules. with 4MiB and 512KB flash along with esptool.py. 
Basically for a completely blank device you'll need to run three commands: 
```
esptool.py write_flash 0x0 frank-0x00000.bin
esptool.py write_flash 0x10000 frank-0x10000.bin
esptool.py write_flash 0x7c000 esp_init_data_default.bin
```

The address of the last one for esp_init_data_default.bin is dependent on flash size. (NodeMCU flash.md has a good table of those): 

-    0x7c000 for 512 kB, modules like most ESP-01, -03, -07 etc.
-    0xfc000 for 1 MB, modules like ESP8285, PSF-A85, some ESP-01, -03 etc.
-    0x1fc000 for 2 MB
-    0x3fc000 for 4 MB, modules like ESP-12E, NodeMCU devkit 1.0, WeMos D1 mini
-    0x7fc000 for 8 MB
-    0xffc000 for 16 MB, modules like WeMos D1 mini pro

GTP95, 2018-07-15:
I updated esptool to the latest version available on pip, now on the nodemcu clone I don't get anymore the "Fatal exception (28)" error, but after reset I don't get any intelligible output. But while doing my tests I found out that there are different versions of esp_init_data_default.bin around, could you please point me to the one you're using, so I can be 100% sure we are using exactly the same binary files? Thank you. I also plan to buy an original nodemcu and a bunch of esp-01 to do some other tests.

GTP95, 2018-07-15:
Good news: I was able to flash all files but the filesystem using the nodemcu flasher on the nodemcu clone, and I got to the point where the serial output is this:
`dld`▒▒;▒$▒f▒▒͡▒size: 4194304 bytes, esp calibration data at sector @ 0x003fb000
env: Environment @ 0x003fa000 size 4096 bytes (4092 real)
=== Current environment ===
sta-mode    dhcp
prompt      frankenstein
default-mode  APSTA
sta-ip      192.168.0.123
sta-mask    255.255.255.0
sta-gw      192.168.0.1
log-level   3
sta-auto    0
ap-ip       192.168.1.1
ap-mask     255.255.255.0
ap-gw       192.168.1.1
hostname    frankenstein
bootdelay   5
telnet-port  23
telnet-autostart  CONFIG_ENV_DEFAULT_TELNET_AUTOSTART
telnet-drop  60
tftp-server  192.168.1.215
tftp-dir    /
tftp-file   antares.com
=== 373/4092 bytes used ===
`
I then tryied to flash the filesystem at 0x003fb000 but it times out every time. Is this the right address?  If yes, could you please elaborate on the upload via tftp?

nekromant, 2018-07-16:
1. tftp is broken right now due to some changes in the SDK.
2. 0x003fb000 is NOT the correct address for filesystem.espfs. Factory calibration data resides there. 

This line should print you the correct address: 
https://github.com/nekromant/esp8266-frankenstein/blob/79094836653ca921f704bde759ba4bacf3ef8fff/src/main.c#L107

Basically the flash layout is something like this

FIRMWARE (IRAM PART) | FIRMWARE ICACHE_FLASH | FILESYSTEM DATA | ENVIRONMENT | ESP CALIBRATION DATA