nekromant/esp8266-frankenstein #100: can't compile firmware

Scratch-b, 2023-01-30:
Hello,
I try to compile this cool firmware, but I have problems.
I install the last SDK from link in your README file (ESP8266_NONOS_SDK-2.1.0-18-g61248df) with

$ make STANDALONE=y

After this I download Frankenstein code, make build directory, "cmake" and "make menuconfig" - no problems.
But when running "make" from build directory:

[  1%] Building C object CMakeFiles/frankenstein.dir/src/newlib-dummies.c.obj
In file included from /home/scratch/ESP8266/esp8266-frankenstein/include/console.h:61:0,
                 from /home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c:46:
/home/scratch/ESP8266/esp8266-frankenstein/include/espmissingincludes.h:13:6: error: conflicting types for 'ets_isr_attach'
 void ets_isr_attach(int intr, void *handler, void *arg);
      ^
In file included from /home/scratch/ESP8266/esp8266-frankenstein/include/espmissingincludes.h:4:0,
                 from /home/scratch/ESP8266/esp8266-frankenstein/include/console.h:61,
                 from /home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c:46:
/home/scratch/ESP8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/ets_sys.h:67:6: note: previous declaration of 'ets_isr_attach' was here
 void ets_isr_attach(int i, ets_isr_t func, void *arg);
      ^
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c: In function '_free_r':
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c:145:2: error: invalid use of void expression
  return (void *) os_free(ptr);
  ^
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c:145:2: warning: 'return' with a value, in function returning void [enabled by default]
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c: In function '_calloc_r':
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c:175:5: warning: implicit declaration of function 'pvPortCalloc' [-Wimplicit-function-declaration]
     return (void *) os_calloc(sz * num_el);
     ^
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c:175:5: warning: 'return' with a value, in function returning void [enabled by default]
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c: At top level:
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c:178:6: warning: conflicting types for built-in function 'calloc' [enabled by default]
 void calloc(size_t sz, int num_el) {
      ^
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c: In function 'calloc':
/home/scratch/ESP8266/esp8266-frankenstein/src/newlib-dummies.c:179:5: warning: 'return' with a value, in function returning void [enabled by default]
     return (void *) os_calloc(sz * num_el);
     ^
make[2]: *** [CMakeFiles/frankenstein.dir/build.make:82: CMakeFiles/frankenstein.dir/src/newlib-dummies.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/frankenstein.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

What is wrong?
Thank you for your help.

Scratch-b, 2023-02-10:
Thank you for your respond. Can you tell me the version of appropriate SDK and  if you know, where I can download it.

nekromant, 2023-02-10:
https://github.com/nekromant/antares/commit/47f1a525d51070d59799db84547719086deb45e0 Looks like 1.0 was the latest update. Back in the day I downloaded The SDK from espressif's site and compiled the toolchain via ct-ng fork for esp. Not sure how much that changed.

closed 2023-02-10 by nekromant